VUnit/vunit #555: VUnit.add_source_files_from_csv() does not allow for all kwargs to be used

GlenNicholls, 2019-10-02:
`VUnit.add_source_files_from_csv()` only passes vhdl_standard kwarg to `Library.add_source_file()`. However, my file gatherer uses `.ahd` file extensions to denote altera, `.xhd` to denote xilinx, and so on. I need the `file_type` available at least, but all of them should also be available.

kraigher, 2019-10-03:
Sure I guess any argument that `add_sources_files` has would make sense to add to `add_sources_files_from_csv`. Are you able to make a pull request to add it yourself?

GlenNicholls, 2019-10-07:
I am, but where I work I need to get #556 resolved first as that is more critical. Once I can figure out all my problems in that issue then I can knock this one out.