VUnit/vunit #559: Do not add builtins by default

umarcor, 2019-10-07:
From https://github.com/VUnit/vunit/pull/507#discussion_r328261564:

> [@kraigher]
> This is really an old mistake that I think we made that I might want to change. It would have been better not to add the builtins by default since it makes options easier and also we do not require a separate VUnit class to be used for Verilog.

> [@umarcor]
> Do you mean to remove compile_builtins from from_argv and from_args? That'd be a breaking change, wouldn't it?

kraigher, 2019-10-12:
I do not think this is urgent to do. But when we see a reason where this change would make our life easier we should do it. Currently I think we can wait.

kraigher, 2019-10-12:
When doing this change we should replace the  `add_builtins` method with an `add_vhdl_builtins` and `add_verilog_builtins` method.

PR #764 (2023-04-23): remove compile_builtins and vunit.verilog
Close #559.

This PR is the complement of #757. There, deprecation are announced. Here the features are removed.
This PR will be kept as a draft, because it is expected to be merged in future releases, not in the next one.
closed 2023-04-23 by eine