VUnit/vunit #562: vunit_restart does not run files from `modelsim.init_files.after_load`

felixn, 2019-10-08:
We have a use-case where we use the `modelsim.init_files.after_load` feature to set up some required simulation parameters (via ModelSim `force` commands in a .do file).

This works fine in batch mode and in `--gui` mode for the first simulation run, but when using `vunit_restart` the after_load files are not re-run (`_vunit_source_init_files_after_load` is not rerun).
`vunit_load` re-runs these files, but it also re-runs vsim - throwing away all changes to the waveform viewer and so on.

Can we add the possibility to re-run the after_load files with vunit_restart?
I know there are cases where re-running the after_load files is not desired (e.g. if they're used to set up signals in the waveform viewer), but maybe as an optional parameter to vunit_restart?

If we can agree on an approach, I can work on implementing it.

kraigher, 2019-10-08:
Maybe there needs to be a new hook that is called `before_run`

felixn, 2019-10-11:
Closed by #564

closed 2019-10-11 by felixn