VUnit/vunit #565: Problem starting simulation run with ModelSim Microsemi 10.2c

zznaki, 2019-10-10:
Hi 
I am having problems using Vunit 4.1  with modelsim I have been using it with GHDL on a mac osx over the last year without a problem but had to move to modelsim on Windows 10 and Python 3.7

When I try to run a testbench from the command line the runner fails whilst creating a a Persistent TCL shell for model sim. It is probably an issue with my configuration but I can not figure out what.

The simulation functions without a problem is I open a gui with a --gui switch

The output from the command line is below
Any help would be greatly appreciated
Regards
Chris

```
>python run.py --verbose main.tb_unit_rampgen*
Re-compile not needed

Running test: main.tb_unit_rampgen.all
Running 1 tests

Starting main.tb_unit_rampgen.all
Output file: vunit_out\test_output\main.tb_unit_rampgen.all_2ed029cda75a082257d43620377f6bfb39d9e490\output.txt
ModelSim> source "C:/Users/…./simulation/vunit/vunit_out/test_output/main.tb_unit_rampgen.all_2ed029cda75a082257d43620377f6bfb39d9e490/modelsim/common.do"
Traceback (most recent call last):
  File "C:\Program Files\Python37\lib\site-packages\vunit\test_runner.py", line 217, in _run_test_suite
    read_output=read_output)
  File "C:\Program Files\Python37\lib\site-packages\vunit\test_list.py", line 101, in run
    test_ok = self._test_case.run(*args, **kwargs)
  File "C:\Program Files\Python37\lib\site-packages\vunit\test_suites.py", line 66, in run
    results = self._run.run(*args, **kwargs)
  File "C:\Program Files\Python37\lib\site-packages\vunit\test_suites.py", line 165, in run
    sim_ok = self._simulate(output_path)
  File "C:\Program Files\Python37\lib\site-packages\vunit\test_suites.py", line 225, in _simulate
    elaborate_only=self._elaborate_only)
  File "C:\Program Files\Python37\lib\site-packages\vunit\vsim_simulator_mixin.py", line 309, in simulate
    return self._run_persistent(common_file_name, load_only=elaborate_only)
  File "C:\Program Files\Python37\lib\site-packages\vunit\vsim_simulator_mixin.py", line 273, in _run_persistent
    if self._persistent_shell.read_bool("failed"):
  File "C:\Program Files\Python37\lib\site-packages\vunit\persistent_tcl_shell.py", line 78, in read_bool
    assert result in ("true", "false")
AssertionError
fail (P=0 S=0 F=1 T=1) main.tb_unit_rampgen.all (0.5 seconds)

```

kraigher, 2019-10-10:
Are you familiar with Python. Can you add a `print(result)` in the persistent_tcl_shell.py line 77 and show me what is the value?

zznaki, 2019-10-10:
It prints: #VUNIT_RETURN

I worked out that I can work around it with -u which avoids the persistent shell but obviously slows things down a bit

kraigher, 2019-10-10:
There should be a transcript in vunit_out/modelsim/. Can you send it?

zznaki, 2019-10-10:
Transcript file relating to first issue. I see it does not like the path with the space in it. Is there a location where I can change the path from "Program Files" to  "Progra~1". Would this solve the problem. 

```
# C:\Program Files\Python37\lib\site-packages\vunit\tcl_read_eval_loop.tcl 
# invalid command name "C:Program"
puts #VUNIT_RETURN
source "C:/..../simulation/vunit/vunit_out/test_output/main.tb_unit_rampgen.all_2ed029cda75a082257d43620377f6bfb39d9e490/modelsim/common.do"
puts #VUNIT_RETURN
set failed [vunit_load]
# vsim -L vunit_lib -L juice -L uart -L feelib -L main -L feelib_em -L singleaxis_em -L feesim -modelsimini C:/Users/cglee/JUICE/j-mag_fpga/work/MAG_FPGA/simulation/vunit/vunit_out/modelsim/modelsim.ini -onfinish stop -quiet -t ps -wlf C:/Users/cglee/JUICE/j-mag_fpga/work/MAG_FPGA/simulation/vunit/vunit_out/test_output/main.tb_unit_rampgen.all_2ed029cda75a082257d43620377f6bfb39d9e490/modelsim/vsim.wlf -g/tb_unit_rampgen/fvunit=True {-g/tb_unit_rampgen/runner_cfg="active python runner : true,enabled_test_cases : ,output path : C::/Users/cglee/JUICE/j-mag_fpga/work/MAG_FPGA/simulation/vunit/vunit_out/test_output/main.tb_unit_rampgen.all_2ed029cda75a082257d43620377f6bfb39d9e490/,tb path : C::/Users/cglee/JUICE/j-mag_fpga/work/MAG_FPGA/stimulus/,use_color : true"} main.tb_unit_rampgen(testbench) 
# //  ModelSim Microsemi 10.2c Jan 15 2014 
# //
# //  Copyright 1991-2013 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //
# false
puts #VUNIT_RETURN
puts #VUNIT_READVAR=${failed}
quit -force -code 0
```

kraigher, 2019-10-10:
That is likely a bug in microsemi vsim executable. VUnit passes the arguments via argv directly and there is no shell tokenization of the arguments. Thus a space in the path should not matter between VUnit and vsim. Probably the vsim executable fails to handle the unquoted string internally. You can do the experiment yourself by calling vsim -c from the command line with a "-do" argument for a file with a space in the path and see if it fails.

zznaki, 2019-10-10:
I copied the common.do from a test_ouput folder
vsim -c -do c:\Users\cglee\hello there\common.do 
Does indeed fail.
Changing the Python Path environmental variables so that Program Files is replaced with Progra~1 fixes the initial problem (thanks). 

The crash still occurs though.

kraigher, 2019-10-10:
The previous output you showed was not a crash but a genuine failure of your test bench.

zznaki, 2019-10-10:
Ah OK sorry - did not look at it carefully and I am Modelsim is new to me.
 Many thanks for your help

closed 2019-10-10 by zznaki