VUnit/vunit #566: No public API for logging to file?

felixn, 2019-10-11:
I couldn't find a way to log to a file in the logging library without using private pats of the API. Am I missing something?

The following works:
```vhdl
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_logging is
  generic (runner_cfg : string);
end entity tb_logging;

architecture tb of tb_logging is
begin
  main : process
    constant log_file_name: string := output_path(runner_cfg) & "schmift.txt";
    variable file_handler: log_handler_t;
    variable logger: logger_t;
  begin
    test_runner_setup(runner, runner_cfg);

    while test_suite loop
      if run("stdout") then
        warning("A warning");
      elsif run("file") then
        file_handler := new_log_handler(log_file_name, format => verbose, use_color => false);
        logger := get_logger("custom");
        set_log_handlers(logger, (0 => file_handler));
        show_all(logger, file_handler);
        warning(logger, "A warning");
      elsif run("both") then
        file_handler := new_log_handler(log_file_name, format => verbose, use_color => false);
        logger := get_logger("custom");
        set_log_handlers(logger, (0 => file_handler, 1 => display_handler));
        show_all(logger, file_handler);
        warning(logger, "A warning");
      end if;
    end loop;

    test_runner_cleanup(runner);
  end process;
end architecture;
```

But the `new_log_handler()` function is in the private section of log_handler_pkg.vhd:
https://github.com/VUnit/vunit/blob/214de988c32018d19be4101667185747b0bb69fe/vunit/vhdl/logging/src/log_handler_pkg.vhd#L58-L81

Is there another way to log to files? Or can we make `new_log_handler()` part of the public API?

1138-4EB, 2019-10-11:
I am not very familiar with *advanced* logging features. Nonetheless, I checked my scripts, and I am using the same approach:

```vhdl
  constant tb : logger_t := get_logger("tb");
  constant file_name : string := output_path(runner_cfg) & "log.csv";
  constant file_handler : log_handler_t := new_log_handler(file_name, format => csv, use_color => false);

  set_log_handlers(tb, (display_handler, file_handler));
  show_all(tb, file_handler);
  show_all(tb, display_handler);

  info(tb, "Init test");
```

Hence, unless told otherwise, I'm ok with moving it. We might just remove the `Private parts not intended for public use` comment. There is no such distinction in the body.

@felixn, do you want to submit a PR?

kraigher, 2019-10-12:
I think just new_log_handler must be moved to public.

felixn, 2019-10-13:
Yeah, I also think that `new_log_handler()` should be moved to public.

But maybe alternatively, we can make logging to files easier? Maybe an optional `file_name` / `to_file` parameter to `get_logger()` could be used to automatically create a new file log handler? Something like the following?
```vhdl
impure function get_logger(
    name : string;
    parent : logger_t := null_logger;
    to_stdout : boolean := true;
    to_file : string := "") return logger_t;
```

PR #570 (2019-10-16): Move new_log_handler to public section and add default parameters
Move new_log_handler to public section and add default parameters (closes #566)
commit c7c580ce (2019-10-16): Move new_log_handler to public section and add default parameters (closes #566)
closed 2019-10-16 by 1138-4EB