VUnit/vunit #569: [BUG] pre-compiled external Xilinx libraries with ghdl do not work properly

GlenNicholls, 2019-10-15:
I followed this guide to compile vivado unisim, unimacro, and secureip vendor libraries in GHDL:

https://github.com/ghdl/ghdl/tree/master/libraries/vendors

This produces a `.cf` file and I use the `add_external_library()` function to add these like so:

```
def add_external_libs(vunit_obj, lib_path):
    for library_name in ["unisim", "unimacro", "unifast", "secureip", "xpm"]:
        print('Adding external library: {}'.format(library_name))
        path = join(lib_path, library_name)
        if exists(path):
            vunit_obj.add_external_library(library_name, path)
        else:
            print('Path does not exist: {}'.format(path))
```

This works properly for Active-HDL and Riveria-PRO, but does not work for GHDL. The paths to the `.cf` have been verified to be correct, but I am not able to figure out why I get the following error:

```
=== Command used: ===
C:\ghdl\ghdl-0.36\bin\ghdl -a --workdir=P:\<dir>\vunit_out\ghdl\libraries\FPGALib --work=FPGALib --std=08 -PP:\<dir>\vunit_out\ghdl\libraries\vunit_lib -PP:\<dir>\vunit_out\ghdl\libraries\osvvm -PC:\ghdl\ghdl-0.36\vendor_sim\xilinx-vivado\v2019p1\unisim\v93 -PC:\ghdl\ghdl-0.36\vendor_sim\xilinx-vivado\v2019p1\unimacro\v93 -PC:\ghdl\ghdl-0.36\vendor_sim\xilinx-vivado\v2019p1\unifast\v93 -PC:\ghdl\ghdl-0.36\vendor_sim\xilinx-vivado\v2019p1\secureip\v93 -PP:\<dir>\vunit_out\ghdl\libraries\TopLib -PP:\<dir>\vunit_out\ghdl\libraries\PMC_CONTROL_Base -PP:\<dir>\vunit_out\ghdl\libraries\PMC_CONTROL -PP:\<dir>\vunit_out\ghdl\libraries\FPGALib -PP:\<dir>\vunit_out\ghdl\libraries\IO_Lib -PP:\<dir>\vunit_out\ghdl\libraries\fft_lib -PP:\<dir>\vunit_out\ghdl\libraries\ModulatorLib --ieee=synopsys -frelaxed-rules P:\<path>\AsyncFifo.xhd

=== Command output: ==
P:\<path>\AsyncFifo.xhd:43:9: cannot find resource library "unisim"
P:\<path>\AsyncFifo.xhd:44:12: unit "vcomponents" not found in library "unisim"
P:\<path>\AsyncFifo.xhd:46:9: cannot find resource library "unimacro"
P:\<path>\AsyncFifo.xhd:47:14: unit "vcomponents" not found in library "unimacro"
```

In my code, I am adding the vendor dependencies like so:

```
library unisim;
use unisim.Vcomponents.all;

library unimacro;
use unimacro.Vcomponents.all;
```

Is there something I need to do besides provide the path to the pre-compiled library in VHDL? I would think that GHDL would recognize this library is provided with the `-P` flag. Have you had success using vendor pre-compiled libraries in GHDL?

kraigher, 2019-10-15:
I personally do not use ghdl with xilinx. I am sure several people in our community do. I would ask in the gitter chat since people doing this already might not read issues looking for ways to help. 

My theory is maybe GHDL expects the parent directory or a subdirectory.

GlenNicholls, 2019-10-15:
Where is the gitter chat located? I've never used that before

LarsAsplund, 2019-10-15:
https://gitter.im/VUnit/vunit

kraigher, 2019-10-15:
Lars gave you the link but it can also be found on our website.

GlenNicholls, 2019-10-15:
This is not a problem with VHDL. To fix my problem, I had to do the following:

In `run.py`:
```
def add_ghdl_external_libs(vunit_obj, lib_path, vhdl_standard='1993'):
    for library_name in ["unisim", "unimacro", "unifast", "secureip", "xpm"]:
        print('Adding external library: {}'.format(library_name))
        path = join(lib_path, library_name)
        if exists(path):
            if vhdl_standard == '1993':
                vunit_obj.add_external_library(library_name, join(path, 'v93'), vhdl_standard=vhdl_standard)
            elif vhdl_standard == '2008':
                vunit_obj.add_external_library(library_name, join(path, 'v08'), vhdl_standard=vhdl_standard)
            else:
                print('Unsupported VHDL standard: {}'.format(vhdl_standard))
        else:
            print('Path does not exist: {}'.format(path))
```

To get all the libs for 1993 and 2008, I had to `cd <ghdl-0.36>/lib/ghdl` and run the following:

```
./vendors/compile-xilinx-vivado.sh --all
./vendors/compile-xilinx-vivado.sh --all --vhdl2008
```

This will compile all the Xilinx vendor libs to `<ghdl-0.36>/lib/ghdl/xilinx-vivado`. However, even though the folders are in the `v93` and `v08` structure GHDL expects, it will ignore the `.cf` if it is for a VHDL version that is not currently being used.

The above python code fixed this problem, but VUnit defaults to a different VHDL version than GHDL, so this will cause problems when VUnit defaults to vhdl2008 and GHDL defaults to vhdl1993.

@1138-4EB do you think this is something that needs to be reported to GHDL? They are external libraries so I'm not sure if their tool should support looking in directories for v08/v93 in the specific case that vendor libraries are being used.

closed 2019-10-15 by GlenNicholls

1138-4EB, 2019-10-15:
> @1138-4EB do you think this is something that needs to be reported to GHDL?

It is the defined and known behaviour of GHDL. See *Note* in https://ghdl.readthedocs.io/en/latest/references/ImplementationOfVHDL.html#vhdl-standards. Hence, it does not need to be reported as a bug. However...

> They are external libraries so I'm not sure if their tool should support looking in directories for v08/v93 in the specific case that vendor libraries are being used.

This is something that produces confussion to new users, specially to those that are used to vendor simulators that support mixing standard versions (almost all of them, I think). Precisely, it was discussed some weeks ago (ghdl/ghdl#908) and docs were updated by @arcturus140, to hopefully make it easier to understand. Some interesting comments from the thread are:

> [1138-4EB]
> @tgingold, I'm thinking that it would be friendlier for users if GHDL emitted a different error message when trying to elaborate a design with a standard version but files were analyzed with a different version. Instead of *'unit not found in library lib'*, something like *'unit not found in library lib for std version XX, but available for std version(s): ... Did you forget --std?'*. Is this feasible or must GHDL ignore libraries for any version other than the selected one?

> [tgingold]
> I think that we must revisit how ghdl handles vhdl revision.  And do like the commercial simulators.

> [1138-4EB]
> In practice, does that mean to make --std an analysis option only?

> [tgingold]
> Might be more complex as files must have the same view of ieee files.

These ideas were commented, but there is actually no open issue to track them. Feel free to open it/them.

Apart from that, I wonder if VUnit's ghdl interface can be enhanced to account for this context. IRRC, subdirs named `v93` and `v08` are generated by GHDL automatically and you cannot modify their names, you can only set the parent (@tgingold, please correct me if I'm wrong). Hence, it might be suitable to have ghdl_interface automatically add the subdir name depending on the standard. I'm trying to think of context where this might be a problem. Otherwise, I'd be good with it. /cc @kraigher

tgingold, 2019-10-16:
> Apart from that, I wonder if VUnit's ghdl interface can be enhanced to account for this context. IRRC, subdirs named v93 and v08 are generated by GHDL automatically and you cannot modify their [...]

No, GHDL never creates directories.  However, it searches libraries within subdirectories of -Pdir