VUnit/vunit #579: Get run CLI args from Results

umarcor, 2019-10-28:
It would be useful if the CLI args that are used to run each test/config were available programmatically (either in `post_run` or through `--export-json`). Something similar to:

``` python
ui.set_sim_option("ghdl.elab_e", True)
ui._args.elaborate = True

def post_func(results):
    for key, val in results.args:
        print(key) # entity.arch.config
        print(val) # list of args (argv to GHDL)

vu.main(post_run=post_func)
```

In https://github.com/VUnit/vunit/pull/568/files#diff-d11203cc5c5e59bf50c8fe619e03e39d, I'm using a modified ghdl_interface to write the args to a txt file, as a workaround.

---

This would allow to indirectly retrieve the version of the standard, as commented in https://github.com/VUnit/vunit/issues/577#issuecomment-546560476.

kraigher, 2019-10-28:
The args should be readable from the VUnit object already. No need to wait to post_run.

umarcor, 2019-10-30:
I tried to get it from VUnit, but I didn't succeed.

I assume I need to execute `_get_command` from `ghdl_interface` (https://github.com/VUnit/vunit/blob/master/vunit/ghdl_interface.py#L226), for each testbench. How do I get from `get_test_benches` (which returns objects of type `vunit.ui.testbench.TestBench`) to `_get_command` without executing `main()`? The *regular* execution flow seems to be:

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/ui/vunit.py#L751-L764

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/ui/vunit.py#L921-L942

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/test_runner.py#L78-L115

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/test_runner.py#L139-L160

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/test_runner.py#L187-L224

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/test_suites.py#L170-L188

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/test_suites.py#L227-L260

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/ghdl_interface.py#L264-L278

https://github.com/VUnit/vunit/blob/895e8c489b414dce83d044ee804c82075f898c3f/vunit/ghdl_interface.py#L226

umarcor, 2019-10-30:
> Oh are you talking about the argv to ghdl?

Yes. The point is that `ghdl_e` allows to build a binary. Therefore, the runtime arguments that VUnit creates are discarded/lost. In order to later execute the binary, either from a Python script or externally, those args are required. It is possible, and sometimes desirable, to set those arguments manually; but it'd be useful to get the ones that VUnit creates. When VUnit is used to generate tests with multiple combinations of generics and/or configurations, it is pointless to repeat all the logic.

> I am not sure we want to expose the GHDL command on the public API. I am not sure we can even promise that it will always be one command even for all simulators.

I think it'd be good to retrieve the sequence of commands that VUnit uses for simulation and the arguments for compilation in a machine-readable format (i.e. not parsing the logs). I'm ok with not adding them to the public API, but to provide guidelines about how to achieve it. Would you be open to merging something like https://github.com/VUnit/vunit/blob/effda9b20f30b3f47ed40c389d993c9724890f32/vunit/ghdl_interface.py#L260-L267 ?

kraigher, 2019-10-30:
If we add a public abstraction of the simulator it must not constrain the implementation of any current or future simulator. Do we want to promise that each simulator will always need one command to run a test?

It seems like yor use case is quite niche and GHDL specific and would be better accommodated by something similar as an args.txt you propose. Maybe make it have some standard format such as json such that string escaping and tokenization does not become a problem.

umarcor, 2019-10-30:
> If we add a public abstraction of the simulator it must not constrain the implementation of any current or future simulator. Do we want to promise that each simulator will always need one command to run a test?

The main motivation to propose exposing it with a mechanism different from `args.txt` was that I thought you would not like the later. In that case, we would not promise that each simulator will need one command, but that the exposed object would contain the number of commands that is required. I.e., the artifact would be a list of lists of strings, instead of a list of strings. Anyway, since you are ok with writing a file to the same path as the simulation binary, no need to worry about the hypothetical public abstraction.
 
> It seems like yor use case is quite niche and GHDL specific and would be better accommodated by something similar as an args.txt you propose. Maybe make it have some standard format such as json such that string escaping and tokenization does not become a problem.

I'll convert it to JSON and I'll propose a PR.

umarcor, 2020-02-20:
Merged as part of #606.

closed 2020-02-20 by umarcor