VUnit/vunit #584: Active-HDL 10.5a not adding package generics

aappl, 2019-11-11:
When using Active-HDL 10.5a as the simulator, the OSVVM packages that contain package generics are not added. The packages missed are "ScoreboardGenericPkg.vhd", "ScoreboardPkg_int.vhd", and "ScoreboardPkg_slv.vhd". When running with 10.5 the files are included.

To reproduce:
- Set environment variables to select Active-HDL 10.5a.
- Add the add_osvvm() call to your run.py.
- Run your run.py file with the -f flag.
- See that the OSVVM files listed do not include the ones stated above.

It looks like the VersionConsumer class in activehdl_interface.py doesn't handle non-digits in the version number.

LarsAsplund, 2019-11-11:
I don't have both versions but does it works for you if you change the identified regexp to `_version_re = re.compile(r"(?P<major>\d+)\.(?P<minor>\d+)[a-zA-Z]?\.\d+\.\d+")`?

aappl, 2019-11-11:
Yes, that does seem to fix it for 10.5a.

LarsAsplund, 2019-11-12:
Good. One more question before we can update the code. Is 10.5a a later release then 10.5 or the other way around?

aappl, 2019-11-12:
10.5a is the later version.

closed 2019-11-23 by LarsAsplund