VUnit/vunit #588: Constraints on generic types

1138-4EB, 2019-11-21:
> https://github.com/VUnit/vunit/pull/557#discussion_r347529855
> vunit does not support real to be passed as generic.

> https://github.com/VUnit/vunit/pull/557#discussion_r349299021
> It's not VUnit but the simulator that sets the limitations on what generic types that are supported. GHDL used to be limited to strings only (to support VUnit runner_cfg).

> https://github.com/VUnit/vunit/pull/557#discussion_r349309165
> Shall we update that constraint? I just tested the following in GHDL and it seems to work:
> 
> ```vhdl
> entity ent is
>  generic (
>     a: string := "hello";
>     b: natural range 0 to 100 := 5;
>     c: integer := -15;
>     d: real := 2.75
>  );
>  end;
> ```
> 
> A proper test to see which types of generics are supported by each simulator might better fit in https://github.com/VHDL/Compliance-Tests. However, in VUnit we can test (and hopefully use) at least those four. Of course, this is out of the scope of this PR.

> https://github.com/VUnit/vunit/pull/557#discussion_r349320189
> I think we should keep it so that we can run the acceptance test on all supported simulators

LarsAsplund, 2019-11-27:
> If we can't find a way to pass plain JSON as generics we need to provide some other safe encoding that can easily be converted back to normal JSON with a to_json function implemented in VHDL. Maybe the easiest way forward, Trying to get things solved at the vendor side will only slow us down and end up in special solutions for each vendor

>Good point. That safe encoding might be base64. It is quite common in web environments and Python provides a package to handle it. Precisely, it is used in https://github.com/VUnit/vunit/pull/568/files#diff-c9290a21e8aaf78614c59a6a9a178576. base64 is also mentioned in many references regarding IP encryption. Unfortunately, I am not aware of any open source VHDL library that allows base64 decoding.

LarsAsplund, 2019-11-27:
Using base32 would be even more safe since it only has alphanumericals. It takes more space but that shouldn't be a concern for our use case. We could even do base16 (hex) for easier implementation. I can have a go at this. I'll probably try to push the VHDL decoding part to the JSON4VHDL project. Any comments @paebbels?

1138-4EB, 2019-11-27:
IIRC, the main limitation of the current implementation of JSON-for-VHDL is that it is synthesisable. For example, the maximum length of the file to be read is hardcoded. Precisely, constant `C_JSONFILE_INDEX_MAX` is used to allocate `variable Stream : STRING(1 to StrLength);`: https://github.com/Paebbels/JSON-for-VHDL/blob/master/vhdl/JSON.pkg.vhdl#L297-L309. For simulation, I believe that it would be possible to do `for i in 0 to integer'high-1 loop` and progressively reallocate `Stream`. However, that would not work for synthesis. This is to say that extending JSON-for-VHDL might require explicitly describing which features are for simulation only.

felixn, 2019-11-27:
> It takes more space but that shouldn't be a concern for our use case.
I'm fine with base32 or base16, however I fear that the different simulator vendors all have different, arbitrary limits on lengths of generics that can be passed in via command line...
The data.json currently used in the tests is something like 272 characters, base64 is 364 chars, base32 is 440 chars, base16 is 554 chars.

1138-4EB, 2019-11-28:
While working on #595, I found https://github.com/VUnit/vunit/blob/master/tests/acceptance/artificial/vhdl/tb_set_generic.vhd. If that test passes in all the supported simulators, generics of type real should be supported. Hence, I am curious about the comments by @dstadelm and @LarsAsplund that are quoted above. In which context do generics of type real fail?

dstadelm, 2019-12-09:
If I use a real value for tb_axi_stream I get:

/usr/local/bin/ghdl:error: unhandled type for generic override of 'g_stall_master'
/usr/local/bin/ghdl:error: error during elaboration

eine, 2019-12-09:
@dstadelm, you are correct. It seems that GHDL does supports real values as generics of a top-level entity, as long as a default value is provided and it is not overwritten through CLI args. That's why: https://github.com/VUnit/vunit/blob/master/tests/acceptance/artificial/vhdl/run.py#L63-L66

PR #595 (2020-02-14): json4vhdl: use base16 encodings
Close #588

This is a proof of concept to encode stringified JSON generics in Python and decode them in VHDL. Example json4vhdl is reworked to handle three separate tests: stringified JSON, stringified and b16encoded JSON, and path to a JSON file.

In the current implementation, a package named `encodings` is added to VUnit. This includes `b16decode` and `b16encode` only, for now. Nonetheless, as commented in #588, this package might be merged into JSON-for-VHDL, instead of being added to VUnit.
closed 2020-02-14 by eine