VUnit/vunit #589: .dat input files not found

haggaie, 2019-11-22:
I'm trying to simulate Verilog modules that are generated automatically using Vivado HLS. Some of these modules refer to auxiliary input files that initialize arrays (using $readmemh). Usually, Vivado copies the files to the current directory of the simulator before running the simulator. Is there a standard way of doing this with vunit?

trailbound, 2019-11-25:
I concur! Would be nice if VUnit would provide a clean, standard way to copy files into the working directory of the simulation run prior to simulation start. Or at least provide the simulation working directory in the pre_config hook.

I've run into this issue as well using a Vivado+Questasim flow. There are multiple file types to load memories or coefficients that Vivado expects to be copied into the working directory of the simulation run prior to simulation. I worked around the issue by using a pre_config hook and adding a relative reference to the output_path in order to copy files into the simulation working directory, but this is ugly. 

The work around goes something like this:
`
from shutil import copyfile

class Configurator:
    def pre_config(self, output_path):
        # Copy files to the working directory of the simulation run.
        copyfile(join(ip_path, "fir_compiler_0/fir_compiler_0.mif"), join(output_path, '../../modelsim/fir_compiler_0.mif'))
        copyfile(join(ip_path, "fir_compiler_0/filter_coef.coe"), join(output_path, '../../modelsim/filter_coef.coe'))
        return True

c = Configurator()

tb = tb_lib.test_bench('demo_tb')
t = tb.test('demo_test')
t.set_pre_config(c.pre_config)
`

1138-4EB, 2019-11-26:
AFAIK, in VUnit there is no concept such as *the working directory of the simulation*. All the paths that are provided to the simulators are absolute paths, so that it is not relevant where is the simulator called from. For example, you can add `print(cmd)` in https://github.com/VUnit/vunit/blob/master/vunit/sim_if/ghdl.py#L292. Executing example `array_axis_vcs` will show:

```
'/usr/local/bin/ghdl'
'--elab-run'
'--std=08'
'--work=lib'
'--workdir=/src/examples/vhdl/array_axis_vcs/vunit_out/ghdl/libraries/lib'
'-P/src/examples/vhdl/array_axis_vcs/vunit_out/ghdl/libraries/vunit_lib'
'-P/src/examples/vhdl/array_axis_vcs/vunit_out/ghdl/libraries/osvvm'
'-P/src/examples/vhdl/array_axis_vcs/vunit_out/ghdl/libraries/lib'
'-o'
'/src/examples/vhdl/array_axis_vcs/vunit_out/test_output/lib.tb_axis_loop.test_260bb5c8c675e898eca5dc9024a4420ede12c0bc/ghdl/tb_axis_loop-tb'
'tb_axis_loop'
'tb'
'-gtb_path=/src/examples/vhdl/array_axis_vcs/src/test/'
'-grunner_cfg=active python runner : true,enabled_test_cases : test,output path : /src/examples/vhdl/array_axis_vcs/vunit_out/test_output/lib.tb_axis_loop.test_260bb5c8c675e898eca5dc9024a4420ede12c0bc/,tb path : /src/examples/vhdl/array_axis_vcs/src/test/,use_color : true', '--assert-level=error']
```

I think that the standard way to handle this would be to use `tb_path` (see http://vunit.github.io/user_guide.html#special-generics-parameters). You would put the files in a location relative to the testbench and *fix* HDL sources to prepend paths with `tb_path`.

@trailbound, note that `join(output_path, '..', '..')` is equal to `join(dirname(__file__), 'vunit_out')`, i.e., a subdir which is a sibling of `run.py`. Hence, it seems that you don't need `pre_config` to copy the `*.mif` and `*.coe` files. You can `mkdir -p vunit_out/modelsim`  and `cp` them before running `run.py`. Of course, you can do it in Python too; the point is that you don't need `output_path` because you are using no specific info besides name `vunit_out`, which you will know if you change the default.

BTW, note that `join(output_path, '../../modelsim/fir_compiler_0.mif')` might not work on Windows. It is better to use `join(output_path, '..', '..', 'modelsim', 'fir_compiler_0.mif')`.

kraigher, 2019-11-26:
See this discussion for a solution: https://github.com/VUnit/vunit/issues/236

haggaie, 2019-11-26:
Thanks everyone.

I prefer not to change the source Verilog files as they are auto-generated from HLS. I used a solution that copies in the pre_config function, even though I agree that passing a tb_path parameter would be cleaner.

closed 2019-11-26 by haggaie

curiousengineer, 2021-01-03:
Is it true that a "proper" method has not yet been created for this in VUnit even though this problem has been going on for some years now?

Ahmad-Zaklouta, 2022-03-01:
> Hence, it seems that you don't need `pre_config` to copy the `*.mif` and `*.coe` files. You can `mkdir -p vunit_out/modelsim` and `cp` them before running `run.py`. Of course, you can do it in [...]

@eine 

Aren't these files suppose to be in the directory of the testcase itself? or is it enough to be in **vunit_out/modelsim** directory?

LarsAsplund, 2022-07-10:
@curiousengineer The proper method with VUnit is to use `pre_config` and copy files to the simulator working directory using the `simulator_output_path` argument. See https://vunit.github.io/py/ui.html#pre-and-post-simulation-hooks

For sourcing TCL scripts before the simulation have a look at https://vunit.github.io/py/opts.html#simulation-options