VUnit/vunit #590: Run testbench with different parameter files

marph91, 2019-11-23:
I asked the question [here](https://stackoverflow.com/questions/58980752/run-same-testbench-with-different-parameter-files-in-vunit) already, but I guess it was the wrong channel. So here is a short summary.

I have a design with a toplevel file, which is parametrizable by `top_parameter.vhd`. Now I want to create multiple testcases with different `top_parameter.vhd` files. Basically, all other files don't change. How could the testcases be structured without duplicating same files (like `run.py` and `tb_top.vhd`) and without compiling the same vhd files, included by the toplevel file, again for each testcase?

kraigher, 2019-11-23:
VUnit does not support modifying the library hierarchy between running tests. It builds upon that there is *one* definition of which files belong in which library. Thus the pattern of compiling in another file for different test runs is not supported. This is a simplification that enables tests to be run in parallel and it matches well with good design practices that work well in for example Sigasi IDE where you want a consistent library mapping.

You have several options in order from (best to worst):
1. Use generics for parameters instead of compiling different files.
1. Use VHDL configuration feature so "switch" which entity is instantiated within another block.
1. Use a work-references in the top level and compile it into a different library for every top_parameter.vhd which also goes in the same library.
1. Add a command line switch to you run.py file to select the top_parameter.vhd.

marph91, 2019-11-23:
Thank you for the explanation and suggestions :)

I tried to elaborate the options for my usecase. For now point 3 is working for me, but I'm considering to rewrite the design in order to accept the parameters as generics.

closed 2019-11-23 by marph91

LarsAsplund, 2019-11-24:
@marph91 Stackoverflow is not the fastest way to get feedback but I do go there and search for questions once in a while. Do you want to tick the box an answer your own question?