VUnit/vunit #592: How many python3 threads can open at the same time(-p switch)

zhajio1988, 2019-11-26:
@go2sh hello, 
could you please check how many python3 threads can open?  if i run with -p 30, total 2000 testcases, i found the simulation is hang. not the testcase issue, if i sim the testcase alone, it can pass.

1138-4EB, 2019-11-26:
Could you please provide more info about the context? What happens with values between `-p 1` and `-p 30`? How many cores does the host have? What is the CPU/RAM usage?

zhajio1988, 2019-11-26:
@1138-4EB  
when I use with -p 30 -q simulation is hang. and can't continue.but the hanged case， if i run them. they will pass.

cat /proc/cpuinfo | grep name | cut -f2 -d: | uniq -c
     72  Intel(R) Xeon(R) Gold 6240 CPU @ 2.60GHz
cat /proc/cpuinfo| grep "physical id"| sort| uniq| wc -l
2
cat /proc/cpuinfo| grep "cpu cores"| uniq
cpu cores	: 18
cat /proc/cpuinfo| grep "processor"| wc -l
72

I will show you later of CPU/RAM usage.

kraigher, 2019-11-26:
What simulator are you using? You might run out of licenses and be blocked because of it.

zhajio1988, 2019-11-26:
synopsys VCS，not come across VCS out of license situation

kraigher, 2019-11-26:
Synopsys VCS is not offcially supported. Are you using a fork?

zhajio1988, 2019-11-26:
i am not using a fork， just local modify.

kraigher, 2019-11-26:
Ok I would consider the possibility that you have a bug in your locally modified code.

zhajio1988, 2019-11-27:
@kraigher all right, i will  check my modified code. thanks.

LarsAsplund, 2019-11-27:
@zhajio1988 Would you consider making your code public. Would allow people to review/spot the bug. Could also be the start of official support for VCS

zhajio1988, 2019-12-02:
@LarsAsplund  hi, my modified scripts is used to run systemverilog/UVM based testbench. 
i will check if i can transplant code to vunit.

LarsAsplund, 2022-07-10:
Closing this due to inactivity

closed 2022-07-10 by LarsAsplund