VUnit/vunit #594: VUnit 4.1.0 breaks VHDL 2008 library settings for GHDL

ttobsen, 2019-11-27:
Since upgrading to VUnit 4.1.0, I've git problems using with GHDL.

For example, I add in my run.py 

```python
test_lib = vu.add_library("test_lib")
test_lib.add_source_files("tests/hdl/*.vhd")
```

I get the following error when running run.py

```
  File "/home/baumannt/devel/workspaceSigasi/vertical-lines-fpn/venv/lib/python3.8/site-packages/vunit/ui.py", line 838, in main
    all_ok = self._main(post_run)
  File "/home/baumannt/devel/workspaceSigasi/vertical-lines-fpn/venv/lib/python3.8/site-packages/vunit/ui.py", line 882, in _main
    all_ok = self._main_run(post_run)
  File "/home/baumannt/devel/workspaceSigasi/vertical-lines-fpn/venv/lib/python3.8/site-packages/vunit/ui.py", line 909, in _main_run
    self._compile(simulator_if)
  File "/home/baumannt/devel/workspaceSigasi/vertical-lines-fpn/venv/lib/python3.8/site-packages/vunit/ui.py", line 1042, in _compile
    simulator_if.compile_project(self._project,
  File "/home/baumannt/devel/workspaceSigasi/vertical-lines-fpn/venv/lib/python3.8/site-packages/vunit/simulator_interface.py", line 171, in compile_project
    self.setup_library_mapping(project)
  File "/home/baumannt/devel/workspaceSigasi/vertical-lines-fpn/venv/lib/python3.8/site-packages/vunit/ghdl_interface.py", line 157, in setup_library_mapping
    raise RuntimeError("GHDL cannot handle mixed VHDL standards, found %r" % list(vhdl_standards))
RuntimeError: GHDL cannot handle mixed VHDL standards, found ['2008', VHDLStandard('2008')]
```

It seems that the default VHDL standard label isn't working anymore.

kraigher, 2019-11-27:
Can you add prints to ghdl_interface.py printing every time a vhdl standard is added to the `set` in setup library mapping? Our automated tests run ghdl and they have worked so it must be related to something local on your side.

ttobsen, 2019-11-27:
Thanks a lot, you're absolutly right. I'm using a selfmade VUnit helper library which adds some VHDL libs using

```python
    drever_lib = vunit.add_library(VHDL_LIBRARY_NAME)
    drever_lib.add_source_files(VHDL_SRC_DIR, vhdl_standard="2008")
```

Removing the vhdl_stabdard parameter fixes the workflow.

So the question is: What is the right syntax to specifiy the VHDL standard? It seems that the default parameter is `VHDLStandard('2008')` and the given `'2008'` parameter isn't resolved to the same result. Therefore the check in `ghdl_interface.py` fails.

kraigher, 2019-11-27:
On public api a plain string should work. The VHDLStandard type is private implemention detail. Are you using any private methods? Maybe you have found a bug?

ttobsen, 2019-11-27:
Ok, some further tests. When using the `vhdl_standard="2008"` in the `add_library` method like

```python
    drever_lib = vunit.add_library(VHDL_LIBRARY_NAME, vhdl_standard="2008")
    drever_lib.add_source_files(VHDL_SRC_DIR)
```

it is also working.

Can it be that the `vhdl_standard` isn't resolved within the `add_source_files` correctly?

kraigher, 2019-11-27:
I think you have found a bug. We have a test gap here it seems. The Public API argument is not converted to the private data type.

kraigher, 2019-11-27:
Which reminds me I miss a static type system in Python. We should really start to use mypy.

ttobsen, 2019-11-27:
> Which reminds me I miss a static type system in Python. We should really start to use mypy.

Yes, the old discussion between static and dynamic typing. But I totally agree with you. :wink:

kraigher, 2019-11-27:
Using mypy I could prove that the bug exists and fix it.

kraigher, 2019-11-27:
It is fixed on master now.

ttobsen, 2019-11-27:
Thanks a lot!

1138-4EB, 2019-11-27:
@ttobsen, shall we close this issue? Or do you want to test it first?

ttobsen, 2019-11-27:
I can test it. Gimme a few minutes and I'll provide some feedback.

ttobsen, 2019-11-27:
Alright, with the 4.2.0 release my example fails and upgrading to the latest master works as expected! Thanks a lot for the fast response and fix, this issue is solved. :smile:

closed 2019-11-27 by 1138-4EB