VUnit/vunit #604: Console report shows passed, but Modelsim output.txt shows failed

alexg-k, 2019-12-02:
When invoking the test below, the console output shows that the test passed, although the test is supposed to fail. When using the option --gui or checking the output.txt of the simulator manually both indicate that the test does fail. Below see my example. The bug occured using vunit-hdl 4.2.0, Python 3.7.3 and QuestaSim -64 10.7a (Modelsim).

run.py:
````
#!/usr/bin/python3

from os.path import join, dirname
from vunit import VUnit

vu = VUnit.from_argv()
vu.add_osvvm()
vu.add_verification_components()

lib = vu.add_library("myWork")
lib.add_source_file("tb_example.vhd")

vu.main()

````

Testbench:
````
library vunit_lib;
context vunit_lib.vunit_context;
context vunit_lib.vc_context;

library ieee;
use ieee.std_logic_1164.all;

entity tb_example is
  generic(
    runner_cfg : string
  );
end entity;

architecture arch of tb_example is

  signal clk : std_logic := '0';
  signal a : real := 1.0;
  signal b : real := 1.0;

begin

  clk <= not clk after 10 ns;

  process
  begin
    test_runner_setup(runner, runner_cfg);

    for i in 0 to 99 loop
      wait for 20 ns;
      b <= b / 2.0;
    end loop;

    test_runner_cleanup(runner);
  end process;
  test_runner_watchdog(runner, 100 ms);

  process
  begin
    wait until rising_edge(clk);
    if run("myTest") then
      check_equal(a, b, max_diff => 0.1);
    end if;
  end process;

end architecture;
````

The console output:
````
Starting myWork.tb_example.myTest
Output file: vunit_out/test_output/myWork.tb_example.myTest_4cbf4a744485b628e0416fcc9cfdef30a50ef69f/output.txt
pass (P=1 S=0 F=0 T=1) myWork.tb_example.myTest (1.0 seconds)

==== Summary ====================================
pass myWork.tb_example.myTest (1.0 seconds)
=================================================
pass 1 of 1
=================================================
Total time was 1.0 seconds
Elapsed time was 1.0 seconds
=================================================
All passed!
````

The output.txt from modelsim:
````
# vsim -modelsimini /home/aklemd/repos/DigitalFilters/development/test/vunit_out/modelsim/modelsim.ini -wlf /home/aklemd/repos/DigitalFilters/development/test/vunit_out/test_output/xil_defaultlib.tb_normalize_step_beta.test_odVarStepsize_table_real_35bc7b5a0e2dbf24ce28761083747d7d330e7a51/modelsim/vsim.wlf -quiet -t ps -onfinish stop xil_defaultlib.tb_normalize_step_beta(arch) -L vunit_lib -L osvvm -L xil_defaultlib -g/tb_normalize_step_beta/runner_cfg="active python runner : true,enabled_test_cases : test_odVarStepsize_table_real,output path : /home/aklemd/repos/DigitalFilters/development/test/vunit_out/test_output/xil_defaultlib.tb_normalize_step_beta.test_odVarStepsize_table_real_35bc7b5a0e2dbf24ce28761083747d7d330e7a51/,tb path : /home/aklemd/repos/DigitalFilters/development/test/tb/,use_color : true" 
# Start time: 08:47:11 on Dec 02,2019
# ** Note: (vsim-3813) Design is being optimized due to module recompilation...
# //  Questa Sim-64
# //  Version 10.7a linux_x86_64 Mar 26 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
Sourcing file /home/aklemd/repos/DigitalFilters/development/test/tb/waves/default.do from modelsim.init_files.before_run
# ** Warning: NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_real
# ** Warning: NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_real
# ** Warning: NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_real
# ** Warning: NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_real
# ** Warning: NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_real
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_aegp
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/sim_table
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/dut_aegp
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ps  Iteration: 0  Instance: /tb_normalize_step_beta/dut_table
# ** Warning: NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
#    Time: 5 ns  Iteration: 0  Instance: /tb_normalize_step_beta/dut_table/divider_table_i
#        145000 ps - check                        -   ERROR - Equality check failed - Got abs (0.000000e+00 - 7.812619e-03) > 1.000000e-06.
# ** Failure: Stop simulation on log level error
#    Time: 145 ns  Iteration: 2  Process: /tb_normalize_step_beta/COMPARATOR_TABLE File: /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/core/src/core_pkg.vhd
# Break in Subprogram core_failure at /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/core/src/core_pkg.vhd line 84
# Stopped at /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/core/src/core_pkg.vhd line 84
# 
# Test Run Failed!
# 
# Stack trace result from 'tb' command
#  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/core/src/core_pkg.vhd 84 return [address 0x7feff0c4c8c4] Subprogram core_failure
# called from  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/logging/src/logger_pkg-body.vhd 746 return [address 0x7feff0c63919] Subprogram decrease_stop_count
# called from  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/logging/src/logger_pkg-body.vhd 741 return [address 0x7feff0c63a35] Subprogram decrease_stop_count
# called from  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/logging/src/logger_pkg-body.vhd 918 return [address 0x7feff0c59f7d] Subprogram log
# called from  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/check/src/checker_pkg-body.vhd 119 return [address 0x7feff0c72c40] Subprogram failing_check
# called from  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/check/src/check.vhd 2302 return [address 0x7feff0c80023] Subprogram check_equal
# called from  /home/aklemd/.local/lib/python3.7/site-packages/vunit/vhdl/check/src/check.vhd 2270 return [address 0x7feff0c7f95e] Subprogram check_equal
# called from  /home/aklemd/repos/DigitalFilters/development/test/tb/tb_normalize_step_beta.vhd 136 return [address 0x7feff16a1ab7] Process COMPARATOR_TABLE
# 
# 
# Surrounding code from 'see' command
#   79 :   begin
#   80 :     if core_failure_is_mocked then
#   81 :       set(core_failure_mock_state, core_failure_called_idx, 1);
#   82 :       reallocate(to_string_ptr(get(core_failure_mock_state, core_failure_message_idx)), msg);
#   83 :     else
# ->84 :       report msg severity failure;
#   85 :     end if;
#   86 :   end;
#   87 : 
#   88 :   procedure check_core_failure(msg : string := "") is
# 
````

kraigher, 2019-12-02:
The problem is that you are not following the VUnit design pattern resulting in undefined behavior.

The `run` call should be between `test_runner_setup` and `test_runner_cleanup` calls.
A single invocation of the `run("test_name")` indicates to VUnit that a new test named "test_name" has started.  Upon the next `run` call VUnit ends the current tests with status passed and starts a new test. It might be possible for VUnit to detect that you are doing something undefined but this detection mechanism has not been added. You are effectively calling run("myTest") several times since the process loops for every rising edge.

Maybe you are looking for `enabled("test_name")` which just returns `true`/`false` if a test with the given name is running at the moment without affecting the VUnit state machine for test execution like the `run` call. 

Note the `run("test_name")` calls are also parsed by VUnit to determine which tests there are in a test bench.

kraigher, 2019-12-02:
@LarsAsplund what do you think about detecting this situation and give an error for misuse of run?

alexg-k, 2019-12-07:
Thank you for pointing out my issue. I am starting just now to use VUnit for testing VHDL components. A warning for this misuse would be appreciated.

Usually when creating testbenches I like to seperate the Stimulus component and the ResponseChecker component into two processes. The Stimulus process determines what input values are tested and when the testbench is finished. The ResponseChecker checks for equality each time the DUT signals a new valid output value is ready. 

I will have a closer look at the `test_suite` loop, `enabled` and `run` functions to see what I need.

eine, 2019-12-07:
@alexg-k, see https://github.com/VUnit/vunit/blob/master/examples/vhdl/array/src/test/tb_sobel_x.vhd or https://github.com/VUnit/vunit/blob/master/examples/vhdl/array_axis_vcs/src/test/tb_axis_loop.vhd for examples of separated stimulus and check processes that fit the design pattern of VUnit.

bradleyharden, 2019-12-11:
@alexg-k, you can model your test benches off of those examples, but I would argue that they are very "RTL" in style. I think that style misses a huge opportunity to improve the structure of your test benches using the VUnit com library:

http://vunit.github.io/com/user_guide.html

If you are new to VUnit, I would suggest taking a look at it. In most cases, it's much easier to work with messages than it is to synchronize yourself.

closed 2019-12-19 by alexg-k