VUnit/vunit #605: Problem with VUNIT_TEST_OUTPUT_PATH_MARGIN and Riviera-Pro

abaebae, 2019-12-05:
I tried to use the `VUNIT_TEST_OUTPUT_PATH_MARGIN` option to reduce the path length. The paths are shortened, but Riviera has a problem with the shortened path. It runs in loop and does not recover. Using `VUNIT_SHORT_TEST_OUTPUT_PATHS` works fine.

kraigher, 2019-12-05:
Setting VUNIT_TEST_OUTPUT_PATH_MARGIN=0 would be equivalent with VUNIT_SHORT_TEST_OUTPUT_PATHS. Could you be more specific what case is not working? What is the name of the test you are trying to run?

abaebae, 2019-12-06:
I did some testing and I can't verify that setting `VUNIT_TEST_OUTPUT_PATH_MARGIN=0` leads to the same path as `VUNIT_SHORT_TEST_OUTPUT_PATHS`. 
To get the same resulst as `VUNIT_SHORT_TEST_OUTPUT_PATHS (full_name = hash_name)` the margin has to be: `margin = max_path - prefix_len - len(hash_name)` . If `margin > max_path - prefix_len - len(hash_name)` the index gets negative and the path gets longer. Maybe a check should warn the user that the margin set is too big or the margin should be limited?
It seems that Riviera has Problems with paths that are too long but still below 260 chars

kraigher, 2019-12-06:
I do not think there is much VUnit can do in this situation. These parameters are for the user to tune the path lengths. The path length reduction works as specified in your case however it was still not enough for Riviera, which VUnit cannot know.

abaebae, 2019-12-09:
I guess you are right

closed 2019-12-09 by abaebae