VUnit/vunit #608: feature request : I want to add a number to the list and run tests on that number

anadigilogic, 2019-12-29:
When executing a test, it is troublesome to execute if the test case name is long. So, I want to be able to run a specific test just by specifying the number.

For example,

python run.py -l

1: project_lib.testsuite.testcase1
2: project_lib.testsuite.testcase2
3: project_lib.testsuite.testcase3

In this case, I want to run the test as follows:

python run 2

kraigher, 2019-12-29:
Wildcard pattern is not enough? In this case you would run *2

anadigilogic, 2019-12-29:
Actually, test cases are not the numbering but the content of the test.

lib.tb_A.config_A.AB Equal
lib.tb_A.config_B.AB Equal
lib.tb_A.config_C.BC Equal
lib.tb_B.config_A.AB Equal
lib.tb_B.config_B.AC Equal
...

In extreme cases, it's a bit tedious to find a pattern that points to the test case in question.

I thought that it would be more intuitive to assign a number when obtaining the list and to be able to execute it.

kraigher, 2019-12-29:
Yes maybe it is useful but you could also add a number to the name of your configurations to make it easier to select tests.

I am on thw fence on this. If more people find this useful I would accept a PR for it. We cannot add every small thing people want especially if it can be solved on the user side unless we make VUnit to bloated.

GlenNicholls, 2020-04-07:
I don't see a reasonable use-case to need this. Copying/pasting after `python run.py -l` is easy enough IMO. Whenever I have failing tests in directories with a lot of tests, I use wildcard patterns if the names are short enough, but copy/paste when I'm testing a bunch of generic configurations and the test names are long. Command prompt is annoying with this so I can understand in that case, but I've been running with bash through vs code and it allows you to use CTRL+C CTRL+V

anadigilogic, 2020-12-22:
I was convinced. thanks a lot.

closed 2020-12-22 by anadigilogic