VUnit/vunit #609: parsing error on python3.8

chatziant, 2020-01-05:
Recently my system (Manjaro Linux) updated to python 3.8 and since then, whenever I try to use Vunit for my tests, the following exception is raised:

> $ python vunit_run_tb.py -v top_lib.tb_test
/home/panos/.local/lib/python3.8/site-packages/vunit/builtins.py:136: Warning: 'array_t' is deprecated and it will removed in future releases;use 'integer_array_t' instead
  warn(arr_deprecation_note, Warning)
Traceback (most recent call last):
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 679, in main
    all_ok = self._main(post_run)
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 725, in _main
    all_ok = self._main_run(post_run)
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 754, in _main_run
    self._compile(simulator_if)
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 898, in _compile
    simulator_if.compile_project(
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/sim_if/__init__.py", line 186, in compile_project
    self.add_simulator_specific(project)
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/sim_if/modelsim.py", line 126, in add_simulator_specific
    mapped_libraries = self._get_mapped_libraries()
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/sim_if/modelsim.py", line 233, in _get_mapped_libraries
    cfg = parse_modelsimini(self._sim_cfg_file_name)
  File "/home/panos/.local/lib/python3.8/site-packages/vunit/sim_if/modelsim.py", line 453, in parse_modelsimini
    cfg.read_file(fptr)
  File "/usr/lib/python3.8/configparser.py", line 718, in read_file
    self._read(f, source)
  File "/usr/lib/python3.8/configparser.py", line 1113, in _read
    raise e
configparser.ParsingError: Source contains parsing errors: '/mnt/LinuxData/Documents/CCSDS-123.0-B-2_HEC/thesis/vunit_out/modelsim/modelsim.ini'
	[line 1970]: 'l   arraymode    - consider rand_mode of unpacked array field independently from its elements\n'
 
Please forgive my ignorance but from what I understand, the modelsim.ini line 1970 isn't parsed correctly, and that's an exception raised on configparser.py. Is there anything I can do to bypass this?
Any help would be appreciated.

felixn, 2020-01-05:
Can you post the offending lines from your modelsim.ini? I bet it looks like this:
```
; Enable/disable non-LRM compliant SystemVerilog constrained-random language extensions.
; Valid extensions are:
l   arraymode    - consider rand_mode of unpacked array field independently from its elements
;   deepcheck    - allow randomize(null) to recursively consider constraints from member rand class handles  
;   forkjoinstab - preserve parent thread random stability when seeding fork/join sub-threads (Default)
;   nonrandstab  - disable seeding of "non-random" class instances (Default)
;   nodist       - interpret 'dist' constraint as 'inside' (ACT only)
;   noorder      - ignore solve/before ordering constraints (ACT only)
;   packrandidx  - allow random index for packed variable in constraint (Default)
;   promotedist  - promote priority of 'dist' constraint if LHS has no solve/before
;   randstruct   - consider all fields of unpacked structs as 'rand'
;   skew         - skew randomize results (ACT only)
; SvRandExtensions = [+|-]<extension>[,[+|-]<extension>*]
```
The line with `arraymode` does not look like a valid line for an ini file, try replacing the `l` with a semicolon.

chatziant, 2020-01-05:
I can confirm that .ini file lines are exactly as you mentioned. 
```
; Enable/disable non-LRM compliant SystemVerilog constrained-random language extensions.
; Valid extensions are:
l   arraymode    - consider rand_mode of unpacked array field independently from its elements
;   deepcheck    - allow randomize(null) to recursively consider constraints from member rand class handles  
;   forkjoinstab - preserve parent thread random stability when seeding fork/join sub-threads (Default)
;   nonrandstab  - disable seeding of "non-random" class instances (Default)
;   nodist       - interpret 'dist' constraint as 'inside' (ACT only)
;   noorder      - ignore solve/before ordering constraints (ACT only)
;   packrandidx  - allow random index for packed variable in constraint (Default)
;   promotedist  - promote priority of 'dist' constraint if LHS has no solve/before
;   randstruct   - consider all fields of unpacked structs as 'rand'
;   skew         - skew randomize results (ACT only)
; SvRandExtensions = [+|-]<extension>[,[+|-]<extension>*]
```
Even if I replace `l` with `;`, the same problem is triggered every time I try to run the vunit_run script.
`modelsim.ini` is produced when the script is executed, so this doesn't exactly solves the problem.

eine, 2020-01-05:
> `modelsim.ini` is produced when the script is executed, so this doesn't exactly solves the problem.

You should be able to provide your own `modelsim.ini`. So, let it create the file for you once, then edit/fix it, last set `VUNIT_MODELSIM_INI` (see http://vunit.github.io/cli.html#simulator-specific-environment-variables). Alternatively, as commented in the docs, you might want to modify it in the source (the installation folder of Modelsim), where it is copied from.

@felixn, any info about how did you know that `l` was the issue? I.e., did you find this problem before or did you guess it by looking at the logs of the first message?

kraigher, 2020-01-05:
VUnit copies the installed modelsim.ini file from on every run do the root problem must be fixed there.

chatziant, 2020-01-06:
After considering what @kraigher mentioned, I figured out that `modelsim.ini` on modelsim installation directory had the same problematic line (line 1970). After editing the file and replacing with a semicolon vunit was able to run the test without complaining.

Thank you all for the help.

closed 2020-01-06 by chatziant