VUnit/vunit #613: Cannot find 'enabled_test_cases'

dsp20, 2020-01-17:
While creating an environment in vunit with System Verilog testbench, my simulation fails with error showing "Internal error: Cannot find 'enabled_test_cases' key".
I checked everything and looks fine for me. Can you guess where could be the problem?
[image]

LarsAsplund, 2020-02-04:
Can you provide a [minimal working example](https://en.wikipedia.org/wiki/Minimal_working_example) for this?

piotrva, 2024-01-15:
I had the same issue. Updating my simulator (ActiveHDL in my case - was working on vesrion 11, updated to version 14) solved this issue. 
After the update I deleted vunit_out folder and restarted the PC.