VUnit/vunit #618: Can't execute Modelsim's set IterationLimit

hcommin, 2020-02-04:
Using VUnit, I often find it necessary to increase Modelsim's IterationLimit beyond its default value. (For example, when checking a large memory).

To do this manually, I can enter `set IterationLimit 20000` in Modelsim. However, I can't find any way of automating this with VUnit, so my automated tests fail.

This is very similar to Issues #223 and #262. However, none of `modelsim.vsim_flags`, `modelsim.init_files.after_load` or `modelsim.init_files.before_run` seem to work for me. I can get a tcl script to execute, but the IterationLimit is somehow always unchanged.

kraigher, 2020-02-04:
You need to declare it as global first.

```tcl
global IterationLimit
set IterationLimit 2000
```

Another alternative is to see if there is a vsim flag to set it and use vsim_flags sim_option

hcommin, 2020-02-05:
Thank you - this solves the issue. I created a new file named _init.do_, containing these lines:

```
global IterationLimit
set IterationLimit 20000
```

Then I added this line to my _run.py_:

`lib.set_sim_option("modelsim.init_files.before_run",["init.do"])`

I would prefer a solution using a `vsim` flag (since then I don't need the extra _init.do_ file). However, I scanned through the Modelsim doc and the closest I could see was the `-do` option, which runs into issue #262.

closed 2020-02-05 by hcommin