VUnit/vunit #619: modelsim.ini parsing error

milosreljin, 2020-02-05:
I tried to make a test project with vunit and ModelSim, but when I run compile option, I get this error:

```
Traceback (most recent call last):
  File "/home/rm/testing/scripts/vunit/vunit/ui/__init__.py", line 690, in main
    all_ok = self._main(post_run)
  File "/home/rm/testing/scripts/vunit/vunit/ui/__init__.py", line 734, in _main
    return self._main_compile_only()
  File "/home/rm/testing/scripts/vunit/vunit/ui/__init__.py", line 873, in _main_compile_only
    self._compile(simulator_if)
  File "/home/rm/testing/scripts/vunit/vunit/ui/__init__.py", line 913, in _compile
    target_files=target_files,
  File "/home/rm/testing/scripts/vunit/vunit/sim_if/__init__.py", line 201, in compile_project
    self.add_simulator_specific(project)
  File "/home/rm/testing/scripts/vunit/vunit/sim_if/modelsim.py", line 116, in add_simulator_specific
    mapped_libraries = self._get_mapped_libraries()
  File "/home/rm/testing/scripts/vunit/vunit/sim_if/modelsim.py", line 223, in _get_mapped_libraries
    cfg = parse_modelsimini(self._sim_cfg_file_name)
  File "/home/rm/testing/scripts/vunit/vunit/sim_if/modelsim.py", line 437, in parse_modelsimini
    cfg.read_file(fptr)
  File "/usr/lib/python3.6/configparser.py", line 718, in read_file
    self._read(f, source)
  File "/usr/lib/python3.6/configparser.py", line 1111, in _read
    raise e
configparser.ParsingError: Source contains parsing errors: '/home/rm/testing/vunit_out/modelsim/modelsim.ini'
	[line 1970]: 'l   arraymode    - consider rand_mode of unpacked array field independently from its elements\n'
```
When I opened modelsim.ini, I saw this:

> ; Enable/disable non-LRM compliant SystemVerilog constrained-random language extensions.
; Valid extensions are:
l   arraymode    - consider rand_mode of unpacked array field independently from its elements
;   deepcheck    - allow randomize(null) to recursively consider constraints from member rand class handles  
;   forkjoinstab - preserve parent thread random stability when seeding fork/join sub-threads (Default)
;   nonrandstab  - disable seeding of "non-random" class instances (Default)
;   nodist       - interpret 'dist' constraint as 'inside' (ACT only)
;   noorder      - ignore solve/before ordering constraints (ACT only)
;   packrandidx  - allow random index for packed variable in constraint (Default)
;   promotedist  - promote priority of 'dist' constraint if LHS has no solve/before
;   randstruct   - consider all fields of unpacked structs as 'rand'
;   skew         - skew randomize results (ACT only)
; SvRandExtensions = [+|-]<extension>[,[+|-]<extension>*]

For some reason, one line starts with letter L instead of ;
Any idea why this happens?

milosreljin, 2020-02-05:
Nevermind, I realized that VUnit just copies modelsim.ini from tool's install path and file was corrupted there. When I fixed it manually in installation dir, compile passed.

closed 2020-02-05 by milosreljin

eine, 2020-02-05:
Actually, this is dup of #609. I'm glad you found out so quick!