VUnit/vunit #625: TypeError: 'PosixPath' object is not iterable

talonmyburgh, 2020-02-13:
Following the vivado example provided for vhdl (found at vunit/examples/vhdl/vivado), I run:

`python generate_vivado_project.py`

which launches vivado, runs without error and exits vivado. Then when I run:

`python run.py`

I get the following error:

`Traceback (most recent call last):
  File "run.py", line 24, in <module>
    VU.add_library("lib").add_source_files(SRC_PATH / "*.vhd")
  File "/home/talon/anaconda3/lib/python3.7/site-packages/vunit/ui/library.py", line 196, in add_source_files
    for pattern_instance in patterns:
TypeError: 'PosixPath' object is not iterable`

I've scoured the mentioned Python file and both the run.py and generate_vivado_project.py file to see why this issue is arising but I can't see it.
I run Ubuntu 18.04 and use Vivado 2019.1

Thanks.

felixn, 2020-02-13:
VUnit was updated to support pathlib paths, and the examples were updated to use them in #612 
So the issue is most likely that you have an older vunit installed (probably from pypi) but trying to run examples from a newer git revision. The issue should occur for all examples, even (for example) examples/vhdl/check/run.py.

Either use the examples from a git revision before the PR was merged, or install vunit from an up-to-date git checkout: https://vunit.github.io/installing.html

eine, 2020-02-13:
As explained by @felixn, there is currently a mismatch between the status of examples in `master` and the lates stable codebase (`v4.3.0`). This can be *fixed* cloning VUnit from `master` and setting PYTHONPATH; or browsing the example of tag v.4.3.0: https://github.com/VUnit/vunit/tree/v4.3.0; apart from other install options as linked by @felixn.

I'm closing this issue, because it is an expected transitient context. Nonetheless, feel free to continue the discussion. Vivado does not have a good VHDL support, and other issues might arise: #209

closed 2020-02-13 by eine

talonmyburgh, 2020-02-14:
Okay, I see my mistake. I've now gone and installed VUnit from the 4.3 release and removed the pip installed vunit etc as @felixn mentioned. Now I get the following error when running the run.py script:
`Compiling standard libraries into /home/talon/Desktop/vunit-4.3.0/examples/vhdl/vivado/vivado_libs/standard ...
Traceback (most recent call last):
  File "run.py", line 32, in <module>
    project_file=join(root, "myproject", "myproject.xpr"),
  File "/home/talon/Desktop/vunit-4.3.0/examples/vhdl/vivado/vivado_util.py", line 30, in add_vivado_ip
    compile_standard_libraries(vunit_obj, standard_library_path)
  File "/home/talon/Desktop/vunit-4.3.0/examples/vhdl/vivado/vivado_util.py", line 46, in compile_standard_libraries
    simname = simulator_class.name
AttributeError: 'NoneType' object has no attribute 'name'
`
A bit of digging and trying to manually run this in ipython leads me to think this is an error in trying to use the Vivado simulator? I see support for XSim is a bit of a struggle at the moment given #209 that you mentioned?
So my question is do I need another simulator outside of Vivado? Or is this error owing to something else?

Thanks and sorry if my questions are a bit noob... I'm new to these tools.

eine, 2020-02-15:
> Okay, I see my mistake. I've now gone and installed VUnit from the 4.3 release and removed the pip installed vunit etc as @felixn mentioned.

Installing release v4.3 is equivalent to using `pip install vunit_hdl`. This is because the latest stable release is precisely v4.3. In order to use it, you need to fetch the corresponding examples. I think that you are actually doing this, because of the paths in the error message. This is just a note to remember that the issue is not which version to use, but a mismatch between the version and the examples.

> A bit of digging and trying to manually run this in ipython leads me to think this is an error in trying to use the Vivado simulator? I see support for XSim is a bit of a struggle at the moment given #209 that you mentioned?

Yes, Vivado's VHDL support is quite limited. Vivado is focused neither on simulation nor on synthesis; and their "market-driven development" seems to prioritise Verilog/SystemVerilog. As a result, not only are many of Vivado's own examples broken in VHDL, but some features of the oldest version of the standard are not supported yet.

@qarlosalberto was the last one trying to get #209 to work. I think he tried not enabling any optional VUnit lib and using VHDL 1993. Still, XSim would crash.

Also, according to https://github.com/VUnit/vunit/issues/209#issuecomment-335524828, my overall suggestion is to stay away from Vivado for simulation.

> So my question is do I need another simulator outside of Vivado? Or is this error owing to something else?

I would strongly suggest to use another simulator and optionally another synthesis tool. For simulation, GHDL is the de facto standard. It is the only FOSS tool with good language support, it is known to be pedantic about it, it is lightweight and fast, etc. Actually, GHDL is the tool we use in CI jobs, as licensing restrictions won't allow us to use any other. This is also true for the VASG, and virtually any project in GitHub/GitLab which does CI of VHDL projects.

Apart from that, Xilinx's tools have traditionally played well with Mentor's solutions. ModelSim's/QuestaSim's evaluation/student versions are very used to work around the limitations of XSim. Moreover, some specific VHDL 2008 features are better supported in Mentor's tools than in GHDL. VUnit itself works well on either of them, but depending on the features in your UUT, GHDL might not work yet.

Regarding Aldec's tools, they seem to have the best and most up to date language support. I know that even some VHDL 2019 features have been tested. Unfortunately, I have not used it myself, so I cannot tell you whether evaluation/student versions exist, or how can those be set up with Vivado.

For synthesis, the context is pretty similar. You can try to synthesise with Vivado directly, or you can use any of the mentioned tools as frontends. With GHDL + yosys you can generate a "plain old VHDL". With Mentor's Precision you can generate an EDIF. Once again, I don't know Aldec's equivalent, but I assume there is one. Nevertheless, since you are likely to use VUnit for simulation, but not for synthesis, you might be good with Vivado's own synthesis features.

> Thanks and sorry if my questions are a bit noob... I'm new to these tools.

It's absolutely ok. As long as you don't want/need to rush, we will walk this path together.