VUnit/vunit #632: After moving to pathlib, searching for modelsim.ini is done in a different directory

suoto, 2020-03-06:
When moving to using pathlib (commit 1af512851ca588a677481b78295c650d3af4a1bd) I believe there's a mistake here

https://github.com/VUnit/vunit/blob/bcdabe4ae13e5e49614f9572232f97c516ca261d/vunit/sim_if/modelsim.py#L72-L73

The previous code searched for `modelsim.ini` one level above the executable:

```python
def has_modelsim_ini(path): 
     return os.path.isfile(join(path, "..", "modelsim.ini")) 
```

A suggested fix would be
```python
def has_modelsim_ini(path):
    return os.path.isfile(str(Path(path).parent / "modelsim.ini"))
```

suoto, 2020-03-06:
If the `vsim` executable can be located, one could use `vmap` to work out where `modelsim.ini` is (it also respects the `MODELSIM` environment variable)
```
$ vmap
Reading /home/souto/modelsim/modeltech/linux_x86_64/../modelsim.ini
"std" maps to directory /home/souto/modelsim/modeltech/linux_x86_64/../std.
"ieee" maps to directory /home/souto/modelsim/modeltech/linux_x86_64/../ieee.
...
```

eine, 2020-03-07:
I applied the suggested fix. Thanks for reporting!

closed 2020-03-07 by eine

bjornfor, 2023-11-07:
>If the vsim executable can be located, one could use vmap to work out where modelsim.ini is (it also respects the MODELSIM environment variable)

Implemented in https://github.com/VUnit/vunit/pull/975. Thanks for the idea!