VUnit/vunit #633: Slow pre_config function

qarlosalberto, 2020-03-12:
I have a test with a pre_config() function really slow (about 2 minutes). When I list the tests VUnit executes the pre_config(). It would be possible to list the tests without no pass the pre_config function?

eine, 2020-03-22:
@qarlosalberto, can you please provide a MWE that shows the current (unwanted) behaviour?

qarlosalberto, 2020-03-23:
**adder.vhd**

```
-- half_adder.vhd

library ieee;
use ieee.std_logic_1164.all;

entity half_adder is
  port (a, b : in std_logic;
        sum, carry : out std_logic
    );
end half_adder;

architecture arch of half_adder is
begin
  sum <= a xor b;
  carry <= a and b;
end arch;
```

**adder_vunit_tb.vhd**
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library src_lib;
--
library vunit_lib;
context vunit_lib.vunit_context;

entity half_adder_tb is
  generic (runner_cfg : string);
end;

architecture bench of half_adder_tb is
  -- Clock period
  constant clk_period : time := 5 ns;
  -- Generics

  -- Ports
  signal a, b : std_logic;
  signal sum, carry : std_logic;
begin

  half_adder_inst : entity src_lib.half_adder
    port map (
      a   => a,
      b   => b,
      sum => sum,
      carry => carry
    );

  main : process
  begin
    test_runner_setup(runner, runner_cfg);
    while test_suite loop
      if run("test_alive") then
        info("Hello world test_alive");
        wait for 100 ns;
        test_runner_cleanup(runner);
        
      elsif run("test_0") then
        info("Hello world test_0");
        wait for 100 ns;
        test_runner_cleanup(runner);
      end if;
    end loop;
  end process main;

end;
```

**half_adder_run.py**
```
# -*- coding: utf-8 -*-
from pathlib import Path
from os.path import join , dirname, abspath
import subprocess
from vunit.sim_if.ghdl import GHDLInterface
from vunit.sim_if.factory import SIMULATOR_FACTORY
from vunit   import VUnit, VUnitCLI

################################################################################
#pre_config func
def pre_config_func():
    """
    Before test.
    """
    print("hello")
#post_check func
def post_check_func():
    """
    After test.
    """
    def post_check(output_path):
        check = True
        return check
    return post_check

################################################################################
#VUnit instance.
ui = VUnit.from_argv()

half_adder_src_lib = ui.add_library("src_lib")
half_adder_src_lib.add_source_files("adder.vhd")

#Add tb sources.
half_adder_tb_lib = ui.add_library("tb_lib")
half_adder_tb_lib.add_source_files("adder_vunit_tb.vhd")

#func checks
tb_generated = half_adder_tb_lib.entity("half_adder_tb")
for test in tb_generated.get_tests():
    test.add_config(name="half_adder", pre_config=pre_config_func(),post_check=post_check_func())

#Run tests.
ui.main()
```

qarlosalberto, 2020-03-23:
```
carlos@carlos-xps:~/repo/example_tests/ghdl$ python3 half_adder_run.py --list --verbose
hello
hello
tb_lib.half_adder_tb.half_adder.test_alive
tb_lib.half_adder_tb.half_adder.test_0
Listed 2 tests
```

In this example in the pre_check () function I just print "hello".
But if the pre_check () function is very long (eg signal generation) to list the tests takes me a lot of time. Does it make sense that to list the tests enter pre_chec ()?

eine, 2020-03-25:
At first sight, I'd say it doesn't. However, I don't know if `pre_check` can be used to generate/modify tests dynamically. ping @LarsAsplund, @kraigher.