VUnit/vunit #636: Stability Check Error after 42 s

jonasexpleo, 2020-03-26:
In a project I have some test that run 1min in simulation time with signals that needs to be stable this long. For this I use the `check_stable()` function. The clock of the simulation is 50MHz.
After round about 42s of simulation time the simulation aborts with the following error:

```
RUNTIME: Fatal Error: RUNTIME_0043 check.vhd (848): Value -2147483648 out of range (0 to 2147483647)
```
[check.vhd line 848](https://github.com/VUnit/vunit/blob/master/vunit/vhdl/check/src/check.vhd#L848)

It seams the code uses a clock count which overrun.

The current workaround is to end the first stability check at 30s and start a new one 2 clock cycle later.

Simulator: Aldec Active-HDL 9.2 (sadly i cannot use a newer Version)

eine, 2020-03-26:
Since integers are limited to 32 bits, I guess this could be fixed by extending the counter to build a 64-bit one from chaining two 32-bit signals/variables. However, it would require to define comparison, addition, etc. https://github.com/VUnit/vunit/pull/568/files#diff-8409a338796aa7c3979226d3d5e256ea seems to be a similar solution (for a different use case). Maybe a general enough "64-bit integer type" can be hosted somewhere. It might also be available already, although I am not aware.

LarsAsplund, 2020-03-26:
We could simply use a 64-bit unsigned to extend that range.

LarsAsplund, 2020-04-17:
@jonasexpleo I made a commit on a branch but I wasn't able to easily create a test since it takes "forever" to get to the 31 bit limit. Better that you review the fix and test on you original problem.

jonasexpleo, 2020-04-20:
I know what you mean with it takes "forever". The test took nearly an hour, but with the branch update_check_stable it worked. Thank you.

commit 89957a9b (2020-04-20): Updated check_stable to handle longer time frames. Closes #636.
closed 2020-04-20 by eine