VUnit/vunit #643: RivieraPro's vlib FileNotFoundError

o-haddad, 2020-04-14:
I am new to VUnit and am trying to run through the simple example given in the documentation page.  I am running on a Windows 10 platform with cygwin64 installed with the Python 3.8 module.  I have Aldec RivieraPro 2019.04 installed.  In order to get VUnit to find my simulator, I set the following env variable:  export VUNIT_RIVIERAPRO_PATH=/cygdrive/c/Aldec/Riviera-PRO-2019.04-x64/bin/vsim.

When trying to execute the run.py example given in the documentation, I get this error.  

`$ python3.8 run.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 726, in main
    all_ok = self._main(post_run)
  File "/usr/local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 772, in _main
    all_ok = self._main_run(post_run)
  File "/usr/local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 801, in _main_run
    self._compile(simulator_if)
  File "/usr/local/lib/python3.8/site-packages/vunit/ui/__init__.py", line 947, in _compile
    simulator_if.compile_project(
  File "/usr/local/lib/python3.8/site-packages/vunit/sim_if/__init__.py", line 210, in compile_project
    self.add_simulator_specific(project)
  File "/usr/local/lib/python3.8/site-packages/vunit/sim_if/rivierapro.py", line 121, in add_simulator_specific
    built_in_libraries = self._get_mapped_libraries(self._builtin_library_cfg)
  File "/usr/local/lib/python3.8/site-packages/vunit/sim_if/rivierapro.py", line 252, in _get_mapped_libraries
    proc = Process(
  File "/usr/local/lib/python3.8/site-packages/vunit/ostools.py", line 117, in __init__
    self._process = subprocess.Popen(
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)

**FileNotFoundError: [Errno 2] No such file or directory: '/cygdrive/c/Aldec/Riviera-PRO-2019.04-x64/bin/vlib'**

Exception ignored in: <function Process.__del__ at 0x6fffffa6d160>
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/vunit/ostools.py", line 242, in __del__
    self.terminate()
  File "/usr/local/lib/python3.8/site-packages/vunit/ostools.py", line 215, in terminate
    if self._process.poll() is None:
AttributeError: 'Process' object has no attribute '_process'
`

vlib is in my path and vlib file does exist.

$ which vlib
/cygdrive/c/Aldec/Riviera-PRO-2019.04-x64/bin/vlib

Any help is much appreciated!'

eine, 2020-04-14:
Hi @o-haddad! I'm not aware of VUnit having being tested on cygwin before. On windows, there seems to be multiple different possibilities to get python (https://www.python.org/downloads/, cygwin, msys2/mingw, WSL, docker...), and not all of them are tested. Would you mind trying it on a different setup? Both https://www.python.org/downloads/ and MSYS2 can be downloaded as a tarball, so that you don't need to do permanent modifications on your host. Once we check that the example, VUnit and RivieraPRO do work on a known environment, we can address the difference in your setup.

I'd say the underlaying issue is that `subprocess.Popen` is not using a shell, but calling a system process directly. Hence, the arguments should be converted from `/cygdrive/c/...` to `C:\...`. VUnit does not do any path manipulation of this kind and cygwin seems not to be kicking in.

o-haddad, 2020-04-16:
Thanks for the quick response.  I installed Python 3.8.2 from python.org, installed VUnit again, and attempted to execute run.py from a DOS shell.  I should also note that my Windows platform is a virtual machine running on MacOs with Parallels.  The Mac OS also has python installed, but I don't think that's interfering.  Anyway, I get an even vaguer error now.

[image]

eine, 2020-04-16:
Where is `VUNIT_RIVIERAPRO_PATH` pointing to?

svenka3, 2023-02-12:
I see the same issue with Windows + WSL + Riviera-PRO. I believe the issue is the EXEs in Windows are named as "vlib.exe/vmap.exe" etc. Whereas rivierapro.py plugin looks for "vlog/vmap". For now I did some hack/fix in rivierapro.py in sim_if

```python
 on_wsl = "microsoft" in platform.uname()[2].lower()
        os_exec_str = str(Path(self._prefix) / "vmap")
        if (on_wsl): 
           os_exec_str += ".exe"

        proc = Process(
            [os_exec_str, library_name, path],
            cwd=str(Path(self._sim_cfg_file_name).parent),
            env=self.get_env(),
        )
```

Hitting different error now as:

```log

=== Command output: ===
ERROR VCP5237 "Undefined class: *base_class." "/home/srini/proj/git_VUNIT/cloned_git_vunit/vunit/vunit/verilog/vunit_pkg.sv" 9  19
```

Will clean-up and report again if there is interest in WSL fix above.