VUnit/vunit #645: Add .vh and .svh to allowed file extensions

nfrancque, 2020-04-15:
svh and vh are common systemverilog and verilog file types.  H stands for header.  We should add them to the allowed list here.  

https://github.com/VUnit/vunit/blob/af7b89d389e26c45797f6c82dc8bf5a695670519/vunit/source_file.py#L362

eine, 2020-04-15:
In fact, VUnit seems to generate and use a `vunit_defines.svh` file. It sounds reasonable to be. However, I know nothing about SV honestly. @LarsAsplund?

nfrancque, 2020-04-16:
You know what, I think I might be wrong about this.  I believe headers are included as preprocessor macros and therefore don't need to be compiled.

Wouldn't hurt anything, but I think I am incorrect to want it compiled.

nfrancque, 2020-04-16:
Interesting caveat to this.  They should not be compiled, but do need to be preprocessed, otherwise my relative includes fail.  Already I have added to vunit that it should append the file's directory to the include path for relative includes, but they fail because the actual location in now in preprocessed dir rather than actual, where there are no .svh files to include.

nfrancque, 2020-04-20:
I ended up having to modify it to 1) allow svh files to be added  2) Not do preprocessing.

With this I am able to compile and run a UVM testbench using vunit.  Svh files shouldn't need to be added, but some need to be because there are external libraries that call their files .svh even though they are not headers.

closed 2020-10-15 by nfrancque

LarsAsplund, 2020-10-16:
@nfrancque Did you see this #296?

nfrancque, 2020-10-16:
I closed the issue because I wasn't sure if the problem mattered to anyone else.

The problem IIRC was when I tried making a UVM testbench compile and run using VUnit, some of Mentor's UVMF stuff was named with .svh, even though in reality they should have been .sv as they were not included headers.  I think the UVMF makefiles compiled these files, but I cannot remember for sure.  

I do remember that my above comment was correct, allowing svh and disabling preprocessing (because the file relationship changes for relative includes when the preprocessed files are copied over, for the svh files are used correctly as include files only) made a UVMF environment compilable and runnable with VUnit approximately following [this](https://www.edaplayground.com/x/2NJb).  Not sure if you're the one who wrote that or not.

I don't think that issue is the same, but my SV knowledge is lacking so defer to you there.  My understanding is the include dirs is for svh files used only as includes, which address the preprocessing problem, but the adding svh files into compilation is a Mentor issue for naming the file wrong.  Correct me if wrong, I no longer have access to it to check.