VUnit/vunit #646: Check Results before test complete

Dhananjay20, 2020-04-23:
I have created Vunit test environment. VHDL Simulation generates Output images. I have code to check the output images with reference images in PYTHON. Is there any way I can run the checking script before saying the test is passed or failed. In post_run I can do that but it happens after test is passed.

eine, 2020-04-23:
@Dhananjay20, have a look at https://vunit.github.io/py/ui.html?highlight=post#pre-and-post-simulation-hooks. I.e., use `post_check` instead of `post_run`.

Dhananjay20, 2020-04-23:
Where should I use post check? Can you give any example for it

eine, 2020-04-23:
See the code example in the same section I linked. Just below the description: https://vunit.github.io/py/ui.html?highlight=post#id4. See also https://vunit.github.io/py/vunit.html?highlight=set_post_check#vunit.ui.testbench.TestBench.set_post_check and https://vunit.github.io/py/vunit.html?highlight=set_post_check#vunit.ui.test.Test.set_post_check.