VUnit/vunit #647: Encrypted VHDL (.vhdp) file support

chatziant, 2020-04-27:
Hello,
After reading a previous [issue](https://github.com/VUnit/vunit/issues/81) for the same problem I am struggling to understand how the `add_external_library` includes my .vhdp files to a VUnit library.

I have read and successfully managed to run the example regarding the inclusion on a vivado IP to VUnit. Having that example as a template, I have done the following:

0. Assume that I have a simple 4-bit counter in the encrypted .vhdp file (count.vhdp) and a top level wrapper in an unecrypted file (count_top.vhd), as well as a testbench (tb_count_top.vhd). 
1. I create a vivado project by runing a .tcl script in batch mode, including only the .vhdp file
2. I compile the all the standard libraries (just in case) in the same manner as the example
3. I add the compiled libaries with `add_external_libraries()` to Vunit project.

As it seems no vhdp files where compiled or included to VUnit. I certainly miss something importatnt to make this work. Could you please elaborate on how to include separate .vhdp files to a VUnit project?

Thanks in advance

jorkrohn, 2023-11-15:
Another workaround is to explicitly specify the file type like this:

`lib.add_source_files("*.vhdp", file_type="vhdl")`

LarsAsplund, 2023-11-19:
@chatziant Is this still relevant? If so, I will re-open. From the information given, I can't say what is wrong. @jorkrohn's solution should also work but I've only managed encrypted files by compiling them separately as the IP provider suggests. IPs tend not to change very often so I save VUnit from dealing with the details of them.

closed 2023-11-19 by LarsAsplund