VUnit/vunit #649: Sigasi - Use VUnit installed in a python venv

oholimoli, 2020-05-04:
How can I add my vunit installation in my virtualenv to my sigasi project.

Atm I get the message:
```

Unexpected VUnit version found: "Traceback (most recent call last):
File "D:\GitLab\aca20\aca20_fpga_19.2\test\run.py", line 15, in <module>
from vunit import VUnit
ModuleNotFoundError: No module named 'vunit'"
```

bartbrosens, 2020-05-06:
Sigasi Studio should be started from within the console where the virtualenv was activated. That way Sigasi Studio will use the path as set in the virtualenv and can find the vunit installation there.

I'm updating the documentation on https://insights.sigasi.com/manual/vunit/ to reflect this.

This issue is not affecting VUnit so can be closed here.

LarsAsplund, 2020-05-06:
Thanks Bart.

closed 2020-05-06 by LarsAsplund