VUnit/vunit #650: Majority of VC tests fail in activeHDL 11.1

GlenNicholls, 2020-05-14:
I am having problems using the AXI VCs in activeHDL 11.1 where I see the the simulator times out from the VUnit watchdog. In ModelSim and GHDL, I do not see the same problem with the AXI VCs. Unfortunately, I have been unable to get an MWE that reproduces this bug to report to Aldec. I ran the VC tests for VUnit, and a majority fail with a bound check failure in `integer_vector_ptr_pkg-body-2002p.vhd`. Not too sure what to make of it.

```
VHDL/Verilog/EDIF/SystemC Simulator build 11.1.263.7352

(c) 1997-2020 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `C:\Users\Glen.Nicholls\Desktop\vunit\vunit\vhdl\verification_components\vunit_out\activehdl\library.cfg', `C:\Users\Glen.Nicholls\Desktop\vunit\vunit\vhdl\verification_components\vunit_out\activehdl\vsimsa.cfg'
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

log "C:\Users\Glen.Nicholls\Desktop\vunit\vunit\vhdl\verification_components\vunit_out\test_output\vunit_lib.tb_wishbone_slavefe04f67a4213a873b46be6534e68a911cd79f237\activehdl\transcript"
log -assert "C:\Users\Glen.Nicholls\Desktop\vunit\vunit\vhdl\verification_components\vunit_out\test_output\vunit_lib.tb_wishbone_slavefe04f67a4213a873b46be6534e68a911cd79f237\activehdl\transcript"
source "C:/Users/Glen.Nicholls/Desktop/vunit/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_wishbone_slavefe04f67a4213a873b46be6534e68a911cd79f237/activehdl/common.tcl"
ELBREAD: Elaboration process.
ELBREAD: Elaboration time 0.4 [s].
KERNEL: Main thread initiated.
KERNEL: Kernel process initialization phase.
ELAB2: Elaboration final pass...
KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\Aldec\Active-HDL-11.1\bin\systf.dll'
VHPI: Loading library 'systf.dll'
ELAB2: Create instances ...
KERNEL: Time resolution set to 1ps.
ELAB2: Create instances complete.
SLP: Started
SLP: Elaboration phase ...
SLP: Elaboration phase ... skipped, nothing to simulate in SLP mode : 0.0 [s]
SLP: Finished : 0.0 [s]
ELAB2: Elaboration final pass complete - time: 0.2 [s].
KERNEL: Kernel process initialization done.
Allocation: Simulator allocated 11152 kB (elbread=1280 elab2=7393 kernel=2477 sdf=0)
KERNEL: ASDB file was created in location C:\Users\Glen.Nicholls\Desktop\vunit\vunit\vhdl\verification_components\vunit_out\activehdl\wave.asdb
set failed [vunit_load]
if {$failed} {quit -code 1}
RUNTIME: Fatal Error: RUNTIME_0047 integer_vector_ptr_pkg-body-2002p.vhd (209): Index 255 out of range (0 to 15).
KERNEL: Time: 0 ps,  Iteration: 1,  Instance: /tb_wishbone_slave,  Process: main_stim.
KERNEL: Stopped at time 0 ps + 1.
VSIM: Error: Fatal error occurred during simulation.
C:/Users/Glen.Nicholls/Desktop/vunit/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_wishbone_slavefe04f67a4213a873b46be6534e68a911cd79f237/activehdl/batch.tcl(4) Script: Break at line 4.
VSIM: Simulation has finished.
```

My tests do not produce this error, but at least this is a start to tracking down what the problem is. If you need, I can also zip up the `vunit_out` or `vunit_out/test_output` directories

umarcor, 2020-05-14:
The line it is complaining about (https://github.com/VUnit/vunit/blob/master/vunit/vhdl/data_types/src/integer_vector_ptr_pkg-body-2002p.vhd#L209) is the one used for every `set` of the internal integer vectors, which are used all over VUnit. I would expect such a fundamental issue to arise in the regular test suite...

Can you print `ref`, `index` and `value`? I'd like to understand if any of the following situations is happening:

- `index` is 255, which is not in the "database".
- `index` is retrieving the wrong vector from the "database", so the size does not correspond.
- `index` is retrieving the expected vector, but `255` is an unexpected value.

GlenNicholls, 2020-05-15:
@umarcor 
[output.txt](https://github.com/VUnit/vunit/files/4631631/output.txt)

I also am seeing that this is not the only failure produced by activeHDL, there are others that don't look related...

>I would expect such a fundamental issue to arise in the regular test suite...

I would too, but I'm suspicious that this is an Aldec problem and not a VUnit problem :/

umarcor, 2020-05-19:
@GlenNicholls, thanks! It seems that somewhere `set(ptr, 255, 1)` is being executed, but `ptr` was allocated with size 16. In this case, the id of `ptr` is 128. By searching `ref=28` in the log:

- Items 1...7 are set to 1.
- Items 4...7 are set to 1.
- Items 1...3 are set to 0.
- Seeting item 255 fails.

The point is to locate where is that call produced. It can be some explicit call in your code, or some incorrect internal call.

You might confirm that `ref=128` is allocated with size 16 by printing `length` right before https://github.com/VUnit/vunit/blob/master/vunit/vhdl/data_types/src/integer_vector_ptr_pkg-body-2002p.vhd#L141. That will hopefully provide some hint to find the caller.

GlenNicholls, 2020-09-22:
Sorry for the delay, I have not had time to really dig into this to help with debugging since it is ultimately ActiveHDL's bug. However, I opened a support request and they have looked into the issue and discovered the problem. They are stating that in ActiveHDL 12.0, this problem will be fixed.