VUnit/vunit #652: VUnit fails without error message when loading array from file

marph91, 2020-05-16:
VUnit version: 4.4.0
GHDL version: GHDL 1.0-dev (v0.37.0-524-g06f3682d) [Dunoon edition]

At my project, I tried to execute a testcase, which includes loading of a raw 1.1 MB file into an `integer_array_t`. The simulation fails without any error. I believe it's caused by the `load_raw()` function. Before, when using `load_csv()`, the test failed without error message too, but only after almost one hour waiting.

The corresponding test is available [here](https://github.com/marph91/pico-png/tree/master/sim/png_encoder). The test case with an 800x480 image is not present, because it always fails. I ran it by adding the following lines to `run.py`:
```
testcases.extend([
    Case("ones", 800, 480, 3, 2, 1, 0, [1 for _ in range(800*480*3)]),
])
```

The output is:
```bash
sim> ./run_all.py unittest.tb_png_encoder.ones_800x480*
/home/martin/.local/lib/python3.6/site-packages/vunit/builtins.py:114: Warning: 'array_t' is deprecated and it will removed in future releases;use 'integer_array_t' instead
  warn(arr_deprecation_note, Warning)
Re-compile not needed

Starting unittest.tb_png_encoder.ones_800x480_row_filter_0_color_2_btype_1
Output file: vunit_out/test_output/unittest.tb_png_encoder.ones_800x480_row_filter_0_color_2_btype_1_a8bbaac4fc428ca862268462c45e3ea19651f618/output.txt
fail (P=0 S=0 F=1 T=1) unittest.tb_png_encoder.ones_800x480_row_filter_0_color_2_btype_1 (1.2 seconds)

==== Summary =============================================================================
fail unittest.tb_png_encoder.ones_800x480_row_filter_0_color_2_btype_1 (1.2 seconds)
==========================================================================================
pass 0 of 1
fail 1 of 1
==========================================================================================
Total time was 1.2 seconds
Elapsed time was 1.3 seconds
==========================================================================================
Some failed!
```

It seems to hit some threshold, since running the test with a 800x400 image is successful.

I'm not sure whether this is a GHDL or VUnit issue. How could it be debugged further? If needed, I can try to provide a MWE.

umarcor, 2020-05-19:
@marph91, see #494 and the issues referenced there. See also the point in https://ghdl.github.io/ghdl-cosim/vhpidirect/mistakes.html about large variables/signals.

marph91, 2020-05-19:
Thanks! Increasing the stack size with pythons [resource module](https://docs.python.org/3/library/resource.html#resource.setrlimit) worked for me.

closed 2020-05-19 by marph91

umarcor, 2020-05-19:
Very interesting... Would you mind sharing a small snippet about how you used it? I think that the comment in ghdl-cosim can be enhanced with an example about setting `ulimit -s unlimited` from Python.

marph91, 2020-05-20:
Sure. I pushed it already: https://github.com/marph91/pico-png/blob/b6a18bb6526dae047362d0cf4b00540c1056658d/sim/run_all.py#L44. Feel free to use it. Alternatively, I can also create a PR.

Actually, the first try was to use `subprocess.run(["ulimit", "-s", "unlimited"], shell=True, check=True)`. It didn't work. I guess, because ulimit modifies this value only for the current shell and subprocesses, but Python was originally invoked from another shell.

umarcor, 2020-05-20:
Please, go ahead. I believe it's enough with copying your second paragraph and the previous comment as a sublist in https://ghdl.github.io/ghdl-cosim/vhpidirect/mistakes.html. For example:

> - There are some corner cases when (...) To remove the limit, set `ulimit -s unlimited`. See ghdl#1112.
>   - In Python (say, when using VUnit or cocotb), the stack size can be increased with `resource` module. However, `subprocess.run(["ulimit", "-s", "unlimited"], shell=True, check=True)` does not work; probably because `ulimit` modifies this value only for the current shell and subprocesses, but Python was originally invoked from another shell.