VUnit/vunit #653: tkinter in run.py breaks vunit_restart in Modelsim

hcommin, 2020-05-19:
I would like to run a simple tcl script from within run.py, so I added these lines:

```
from tkinter import Tcl

tcl = Tcl()
tcl.eval('puts [info patchlevel]') # Print tcl version number
tcl.eval('source MyScript.tcl')
```

This seems to work correctly. However, if I then try to run `vunit_restart` from within Modelsim, I get an error containing:

> C:/modeltech_pe_10.7b/win32pe/../tcl/tcl8.6/init.tcl: version conflict for package "Tcl": have 8.6.9, need exactly 8.6.5

Sure enough, `puts [info patchlevel]` reports "8.6.9" from run.py and "8.6.5" from Modelsim.

Is there a simple way I can get everything to agree?

eine, 2020-05-19:
@hcommin, you might use `modelsim.init_files` options to source your script. See http://vunit.github.io/py/opts.html#simulation-options

hcommin, 2020-05-19:
@eine I need to return variables from the tcl script to run.py (which I didn't show in my example above). For example:

`os.environ["VUNIT_MODELSIM_PATH"] = tcl.eval('return $ModelsimPath');`

Am I right to think this can't be done via `modelsim.init_files`?

eine, 2020-05-19:
Can you please provide a MWE? I'm not understanding when and where you want to execute the tcl script. This second issue seems unrelated to the one above. I'm not sure you can set VUNIT_MODELSIM_PATH from `run.py`. Some months ago there were some enhancements to delay the detection of the simulator, but I don't remember the details ATM.

hcommin, 2020-05-19:
Here is a very small working example.

**run.py**
```
from os.path import join, dirname
from vunit import VUnit
from tkinter import Tcl
import os

# Get variable from tcl and print it
tcl = Tcl()
tcl.eval('source Script.tcl')
TclVar = tcl.eval('return $Var')
print('Tcl script returned: ' + TclVar)

# Compile testbench
ui = VUnit.from_argv()
lib = ui.add_library("lib")
lib.add_source_files("tb.vhd")
ui.main()
```

**Script.tcl**
`set Var "Hello world!"`

**tb.vhd**
```
library vunit_lib;
context vunit_lib.vunit_context;

entity tb is
    generic (
        runner_cfg              : string
    );
end entity tb;

architecture rtl of tb is
begin
    
    process
    begin
        test_runner_setup(runner, runner_cfg);
        while test_suite loop
            if run("test0") then
                print("This is a test.");
            end if;
        end loop;
        test_runner_cleanup(runner);
    end process;

end architecture rtl;
```

I then execute `python run.py --gui`, which launches Modelsim. In Modelsim, I execute `vunit_restart`, which produces the error.

eine, 2020-05-19:
I assume that the same MWE works without `tkinter`. I.e., executing `vunit_restart` is NOT broken on the system.

I'd guess that tkinter is a wrapper that loads a shared library, but it is not unloaded before executing `vu.main()`. Because it is not unloaded, Modesim tries to use it, instead of triggering its own procedure. Does it work if the tkinter part is written in a separate file and executed through `check_call`?

hcommin, 2020-05-19:
Executing through `check_call` doesn't appear to fix the issue. I moved these lines out of run.py and into a new sub.py script:

```
from tkinter import Tcl

# Get variable from tcl and print it
tcl = Tcl()
tcl.eval('source Script.tcl')
TclVar = tcl.eval('return $Var')
print('Tcl script returned: ' + TclVar)
```
Without these lines in run.py, `vunit_restart` is **not** broken (and the "Hello world!" variable isn't printed). I then add these lines to run.py:

```
import subprocess

subprocess.check_call('python sub.py');
```

and this leads to the same tcl version error as before.

eine, 2020-05-19:
Then, I think you might need to guess how to explicitly unload/undo whichever modification that importing tkinter produces in the set of loaded shared libs.

It seems that Tcl wraps Tk (https://github.com/python/cpython/blob/3.8/Lib/tkinter/__init__.py#L2362), which has method `destroy`. Calling `tcl.destroy()` might work.

hcommin, 2020-05-19:
My python isn't strong, but as far could understand, the most hopeful method at that link is `quit`. Calling `tcl.quit()` appears to run fine, but doesn't solve the issue.

I searched around about more generally unloading modules, but was surprised that I couldn't find any obvious way of cleaning up. I guess this isn't really a VUnit issue, but it's really going to be a shame if I can't get this to work. My current workaround is hideous.

eine, 2020-05-19:
Unfortunate... As a workaround, you can use some intermediate file (say JSON) and execute both scripts independently:

```sh
python tkscript.py
python run.py
```

Writing/reading JSON is easy in Python, but you need to handle the slightly increased complexity.

GlenNicholls, 2020-07-14:
@hcommin What exactly are you trying to do? I've done a decent amount of work with creating and sourcing TCL scripts in simulators without tkinter.

hcommin, 2020-07-14:
Do you mean what exactly was in the TCL script? I think I was unable to even get as far as "hello world!" because of the version conflict mentioned in my original post. Can you offer any pointers?

GlenNicholls, 2020-07-14:
>I think I was unable to even get as far as "hello world!" because of the version conflict mentioned in my original post. 

I mean, why do you feel the need to use Tkinter in the first place? What is your use case? What information from TCL are you trying to gather in your `run.py` scripts?

>Can you offer any pointers?

Once I understand your answer to the above questions, I will be able to give you some suggestions. While your MWE is useful for understanding that exact code doesn't work, I'm trying to see if there is some XY problem here that needs more clarification before exploring if issues with tkinter are even worth fixing.

hcommin, 2020-07-15:
I think that when I asked this question I was working on a project for a customer who had various tcl scripts for setting up their environment. One of those scripts defined paths to 3rd party tools, such as Modelsim and Vivado. I thought it would be neat if I could source their existing tcl scripts in my run.py. That way, I could set up the VUnit simulator path and (if required) automatically generate vendor-specific simulation models.

But I was also interested in whether tcl is more generally accessible from run.py. We have a lot of ready-made scripts written in tcl, so I thought it could be a time-saver in future if it could be used seamlessly.

Thanks - I will take a look at #665.