VUnit/vunit #654: when running vunit from a subst drive, runner.py line 154 causes a failure (v4.4.0 / python 3.8.3))

olafvandenberg, 2020-05-20:
when starting a simulation from a subst location on windows, runner.py line 154 calls relpath which 
in turn calls ntpath.py leading to the following:

Traceback (most recent call last):
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\site-packages\vunit_hdl-4.4.0-py3.8.egg\vunit\ui\__init__.py", line 726, in main
    all_ok = self._main(post_run)
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\site-packages\vunit_hdl-4.4.0-py3.8.egg\vunit\ui\__init__.py", line 772, in _main
    all_ok = self._main_run(post_run)
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\site-packages\vunit_hdl-4.4.0-py3.8.egg\vunit\ui\__init__.py", line 808, in _main_run
    self._run_test(test_list, report)
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\site-packages\vunit_hdl-4.4.0-py3.8.egg\vunit\ui\__init__.py", line 988, in _run_test
    runner.run(test_cases)
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\site-packages\vunit_hdl-4.4.0-py3.8.egg\vunit\test\runner.py", line 120, in run
    self._run_thread(write_stdout, scheduler, num_tests, True)
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\site-packages\vunit_hdl-4.4.0-py3.8.egg\vunit\test\runner.py", line 154, in _run_thread
    print("Output file: %s" % relpath(output_file_name))
  File "C:\Users\mss7pk\AppData\Local\Programs\Python\Python38-32\lib\ntpath.py", line 703, in relpath
    raise ValueError("path is on mount %r, start on mount %r" % (
ValueError: path is on mount 'D:', start on mount 'W:'

eine, 2020-05-20:
Hi @olafvandenberg! Is this the only issue you have when running the simulation from a subst location? `runner.py` was modified some months ago in order to use `Path` from `pathlib`. I didn't modify `relpath` because of the note in https://docs.python.org/3/library/pathlib.html#correspondence-to-tools-in-the-os-module:

> **Note:** Although `os.path.relpath()` and `PurePath.relative_to()` have some overlapping use-cases, their semantics differ enough to warrant not considering them equivalent.

However, since this seems to just print some info, I think we would also be ok with showing `output_file_name` as is. Do you mind trying `relative_to()` before considering to not show a relative path?

/cc @LarsAsplund

davidmedinasigasi, 2020-06-22:
This issue is also present for some users that use the Sigasi/VUnit integration and have VUnit installed in a different drive.

eine, 2020-06-26:
@davidmedinasigasi, can you please try using `relative_to()` in https://github.com/VUnit/vunit/blob/master/vunit/test/runner.py#L154, instead of `relpath`, and let us know if that works for you?

eine, 2020-06-29:
@olafvandenberg, after merging @davidmedinasigasi's fix, `master` should work for you. Can you please confirm?

closed 2020-06-29 by eine

olafvandenberg, 2020-07-01:
Hi, sorry for the late response, I can confirm that the fix works for me as well..

PR #661 (2020-06-29): do not use 'relpath' when printing output file
This was causing issues when trying to output a XML
file in a different drive than the projects.

Close #654