VUnit/vunit #655: Testbench ghdl cannot open file error

talonmyburgh, 2020-05-20:
Hi all.

VUnit: 4.4.0
OS: Ubuntu 18.04 LTS.
GHDL: GHDL 0.37-dev (v0.36-1612-g93922174-dirty) [Dunoon edition]
            Compiled with GNAT Version: 7.5.0
            llvm code generator
            Written by Tristan Gingold.

This is less of an error I've located in VUnit and more of a plea for help with a testbench that works classically (i.e. using regular VHDL with asserts etc) but not when I switch asserts to 'checks' and run using VUnit with ghdl-llvm as the simulator it fails with the following error:

```
Starting casper_adder_lib.common_add_sub_tb_tb.all
Output file: vunit_out/test_output/casper_adder_lib.common_add_sub_tb_tb.all_b3243b3521fc966158cf78e5e4b3f97953805cfb/output.txt
../../src/ieee2008/numeric_std-body.vhdl:3062:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3062:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_adder/vunit_out/test_output/casper_adder_lib.common_add_sub_tb_tb.all_b3243b3521fc966158cf78e5e4b3f97953805cfb/ghdl/common_add_sub_tb_tb-tb:error: cannot open file "/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_adder/vunit_out/test_output/casper_adder_lib.common_add_sub_tb_tb.all_b3243b3521fc966158cf78e5e4b3f97953805cfb/vunit_results"
in process .common_add_sub_tb_tb(tb).u_add_5_6@common_add_sub_tb(tb).p_in_stimuli
/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_adder/vunit_out/test_output/casper_adder_lib.common_add_sub_tb_tb.all_b3243b3521fc966158cf78e5e4b3f97953805cfb/ghdl/common_add_sub_tb_tb-tb:error: simulation failed
fail (P=1 S=0 F=1 T=2) casper_adder_lib.common_add_sub_tb_tb.all (0.3 seconds)

==== Summary =====================================================
pass casper_adder_lib.common_add_sub_tb.all    (0.3 seconds)
fail casper_adder_lib.common_add_sub_tb_tb.all (0.3 seconds)
==================================================================
pass 1 of 2
fail 1 of 2
==================================================================
Total time was 0.7 seconds
Elapsed time was 0.7 seconds
==================================================================
Some failed!
```

I've checked the file permissions on the file it fails to open and it ought to be able to open it.

I've attached the 3 files the testbench casper_common_add_sub_tb_tb runs multiple instances of the casper_common_add_sub_tb testbench. I feel that the error may be arising due to the passing of the runner_cfg string between test benches but I don't quite know how else to do it.

I am certain that the error does not arise from the used libraries common_pkg_lib and common_components_lib as other modules using these libraries are run by VUnit fine.

Appreciate the help (in advance).

Talon.

[casper_common_add_sub.txt](https://github.com/VUnit/vunit/files/4656026/casper_common_add_sub.txt)
[casper_common_add_sub_tb.txt](https://github.com/VUnit/vunit/files/4656027/casper_common_add_sub_tb.txt)
[casper_common_add_sub_tb_tb.txt](https://github.com/VUnit/vunit/files/4656028/casper_common_add_sub_tb_tb.txt)

eine, 2020-05-20:
Hi Talon! I feel that you are trying to use VUnit in an unidiomatic way. As you suggested, passing `runner_cfg` doesn't feel as good idea. Not because you pass it, but because you have multiple concurrent `test_runner_setup` and `test_runner_cleanup`, which will race with each other. I believe that the architecture of VUnit expects those procedures to be used once only.

Precisely, `casper_common_add_sub_tb_tb` feels like it should NOT be written in VHDL, but either in a shell script or in the `run.py` script instead. You are instantiating the same testbench 8 times, with different generics. See example [Generating tests](http://vunit.github.io/examples.html#id10): https://github.com/VUnit/vunit/blob/master/examples/vhdl/generate_tests/run.py.

You can also have a look at [Composite generics](http://vunit.github.io/examples.html#id12) and [JSON-for-VHDL](http://vunit.github.io/examples.html#id4), for other alternatives to pass sets of top-level generics.

talonmyburgh, 2020-05-21:
Thanks, @eine. It was not a very explanatory error. I've been trying to take an existing VHDL code base and alter all the test benches to operate with VUnit (so as to introduce a better form of continuous integration). Appreciate the help, I will look at the alternatives.

closed 2020-05-21 by talonmyburgh

eine, 2020-05-21:
@talonmyburgh, feel free to ask either here or in the [chat](https://gitter.im/VUnit/vunit) about specific issues you might find while adapting `casper_common_add_sub_tb_tb` to `run.py`. One of the main purposes of VUnit is to facilitate continuous integration and testing of HDL design. In fact, Python test runner features of VUnit are independent from most of the provided VHDL libs.