VUnit/vunit #659: Different simulators for individual Testbenches/Testcases

KlewinS, 2020-06-19:
In our company we have only a limited number of (real) Modelsim licenses. Therefore we are using the free Modelsim version which is shipped together with Quartus for most of the testing.

Since the free version is much slower than the other one, it would be nice if one could configure a differen simulator (the binary of the one to be used) for each individual testcases, or at least for the individual testbenches. In this way one could use the faster one for the more extensive test and the slower one for all other cases.

Of course one could also think of a use case where you could simulate some of the cases with e.g. Modelsim and others with GHDL or any other simulator.

eine, 2020-06-26:
@KlewinS, I think that you can already do this by using environment variables and filtering the tests to be run. Instead of using "global" envvars, set them for each call to VUnit. Once of the features I most like about VUnit is precisely how easily it allows to change from one simulator to another.

LarsAsplund, 2020-06-27:
@KlewinS The environment variables to use would be VUNIT_SIMULATOR to select simulator and then optionally VUNIT_<SIMULATOR_NAME>_PATH to point to the correct bin directory

KlewinS, 2020-06-27:
thanks @eine and @LarsAsplund for pointing to those environment variables. I will try to find a way to set them correctly in my run.py (under windows).

eine, 2020-06-27:
@KlewinS, which terminal do you use on windows?

KlewinS, 2020-06-29:
@eine, usually the default one, but I could also switch to PowerShell. But what I'm using most is the "VUnit Test Explorer"-Plugin for Visual Studio Code. And here, unfortunately, I can't tell what is used in the background (but I woudl guess PowerShell).

eine, 2020-06-29:
@KlewinS, I asked because I use MSYS2 (MINGW64). I use it as the default terminal in VSCode too. Hence, envvars can be defined in the CLI. However, I don't think `VUnit Test Explorer` supports it yet. I think it calls python directly, without intermediate shell.

GlenNicholls, 2020-07-09:
@KlewinS Reference https://github.com/VUnit/vunit/blob/b34a4f3c341f2b4175db678ac7647c32ce48bfae/tests/acceptance/test_external_run_scripts.py#L292 as this shows how you can set or add environment variables. If you use Pytest, it is pretty easy to modify this on a test-by-test or group of test basis.

KlewinS, 2020-07-15:
Thanks for all the nice suggestons for workarounds. I will try to implement one to solve my issue for the moment.

Nevertheless, I wouls still ask you to implement this functionality directly into the python interface. This would be a much clearer solution.

GlenNicholls, 2020-07-15:
>Thanks for all the nice suggestions for workarounds. I will try to implement one to solve my issue for the moment.

Please report back on the solution you choose. I'm going to be working on https://github.com/VUnit/vunit/issues/663 soon and information like this will be valuable to set up examples for users who have similar use-cases.

>Nevertheless, I would still ask you to implement this functionality directly into the python interface. This would be a much clearer solution.

I'm kind of split on this suggestion. I made the same suggestion in the VUnit Gitter some months back and heard some arguments against it, all of which I agree with now. VUnit is an abstract Python interface that serves as many users and use-cases as possible, adding stuff like this to the interface creates clutter and doesn't really provide a benefit to _most_ users. While I like the idea of adding this directly, it would require adding a command-line flag and/or adding Python code (and you'd have to constantly change the Python code). The flag IMO is the cleanest and easiest solution to change when you need to change simulators, but it would not be ideal for different simulator versions of the same simulator because you'd have to type or hard-code the path to the executable. The latter is where I find clutter and lack of value. I think their choice for environment variables strikes a healthy medium between serving as many use-cases as possible while also allowing something that is easy to change from the command line or in Python code.

Like I said, using Pytest and/or setting aliases in `.bashrc` is the best solution since you don't need to constantly type out the path to the executable. Environment variables can be set in your Python code quite easily, so this request doesn't seem like a barrier to use VUnit as the current implementation is sufficient in my eyes. Both VUnit and Pytest have an API for customizing command line arguments, so again, I do not see the barrier. In the case of changing simulator versions, this can be documented in a place where your colleagues can easily access the information. In my case, I put information like this on confluence along with requirements for where to put certain installs/downloads to ensure our VUnit wrapper can add everything it needs to without requiring users to modify any underlying code. There are limitations obviously, but creating clear documentation and instructions ensures users follow the correct methods when setting up their environment for the first time.

These are just my opinions after desiring the same feature and finding a different solution. Again I am not 100% disagreeing with you as I still think simulator selection should be a command-line option and/or Python method. However, I do not think selecting between different simulator versions should be in the VUnit interface as it is far too implementation-specific. I'd be interested in hearing @LarsAsplund thoughts on the matter.

eine, 2020-07-15:
I fully agree with @GlenNicholls. I would not say that I don't want this feature, but (as in #662) I am unsure about the complexity of maintaining a per-test granularity paying off. Filtering tests by name using wildcards is already a quite powerful solution if combined with clear documentation and naming guidelines (which are desirable anyway). Such approach allows to define any external constraint (such as selecting the simulator, setting timeouts, importing custom libs/headers, doing project management -say tsfpga, edalize-...) in the `run.py` without extending the API.

Overall, VUnit does already have many features and the core team is very small. We want to modularize and bring more people who want to take care of specific features or sets of features. In this sense, much of the effort during the last year has gone into refactoring, deprecating Python 2, updating CI, documentation, etc. The next big change is to distribute verification components (#520). Coherently, I believe that all the features which can be conceived as plugins/wrappers around `run.py`, should be implemented as such (e.g. integration with cocotb). Then, some of those could be moved to specific repos, should they be large enough. Of course, sometimes modifying the internals is a requirement. But I think that's not the case here.

I believe we'd better focus on adding examples such as #663, which allow users too achieve equivalent results albeit not using exactly the same implemention they had thought about. @KlewinS, don't take wrong, we've all been there and your request is 100% legit. However, after we get experience with VUnit, multiple pieces and annoyances start to match each other and fit together. Unfortunately, it is hard to communicate these unless you "fight" with the features yourself.