VUnit/vunit #662: Add watchdog to Python API

GlenNicholls, 2020-06-26:
The VHDL watchdog can hang in certain situations when the only thing advancing are delta cycles. This is most prominent when using the communication incorrectly or doing something funky in the VHDL. One nice feature would be to add a watchdog to the Python API as well to catch these situations. This in my mind is an important feature for CI where a test could hang forever until the user or OS kills the process. 

The other problem I've found with these situations is that `CTRL+C` doesn't always work. It will "kill" the test, but there may still be a background process running. Usually killing the Python process manually works, but ActiveHDL sometimes will keep running even after this. I don't think a watchdog in the Python API would fix this problem in ActiveHDL, but it would certainly help CI.

eine, 2020-06-26:
I found the issue with `Ctrl+C` to happen when working with GHDL and doing funky things too. As you say, sometimes `Ctrl+C` once works, sometimes it needs to be triggered twice, sometimes it exists but the simulator is still running in the background and the process needs to be found and killed. Luckily, I think it happens in quite rare situations when the user is already aware of on the limit. Nevertheless, I agree that a more robust solution would be desirable.

My point is that I don't know whether this should be part of the API. Having it built-in would allow to set a watchdog per test. However, we might also handle this as an external script that sets a watchdog for a VUnit run (including include one or multiple tests). The advantage of doing it external is that it'd be really short (5-15 lines) and it could be used regardless of VUnit (e.g., when executing GHDL, ActiveHDL, ModelSim... in other contexts/standalone). Moreover, the "external" script could "query" VUnit with `-l`, and then run all the tests automatically with separate watchdogs. Such script could be contributed to [VUnit/vunit_action](https://github.com/VUnit/vunit_action) (see #656).

Please, feel free to argue otherwise ;)

FTR, in MSYS2 all the update-related processes are terminated with https://github.com/msys2/setup-msys2/blob/master/main.js#L100.

GlenNicholls, 2020-06-26:
>Luckily, I think it happens in quite rare situations when the user is already aware of being on the limit. Nevertheless, I agree that a more robust solution would be desirable.

I agree completely, after the first few times this happened I noticed it and know when I need to kill the process. From my experience, it happens most often when using multiple threads to run tests. However, it isn't that big of a productivity problem since it's quick to fix the problem. Inconvenient, sure, but not critical by any means!

>However, we might also handle this as an external script that sets a watchdog for a VUnit run (including one or multiple tests). The advantage of doing it external is that it'd be really short (5-15 lines) and it could be used regardless of VUnit (e.g., when executing GHDL, ActiveHDL, ModelSim... in other contexts/standalone).

I guess I don't see the value in making an external script as this is something that has to be run manually. Obviously with CI this is automatic, but for running tests manually I could see it being tedious to incorporate. 

My problem with an external watchdog for the group of tests are: the test time can vary drastically with system load, adding more testbenches/parameterized tests can exceed this limit, or you're spending development time externally to the API to set something up to work on a test-by-test basis. The other problem I forsee for a group of tests is that you would likely be killing many tests where only one/few tests expose the test hanging, but the rest should still be run. Both of these cases could be accomplished in the API. My largest opposition to this is that the VHDL run library offers a watchdog, but it isn't perfect and I feel that it seems most appropriate to have a bullet-proof watchdog in the Python API that will ensure the VHDL test never hangs indefinitely.

eine, 2020-06-26:
> My largest opposition to this is that the VHDL run library offers a watchdog, but it isn't perfect and I feel that it seems most appropriate to have a bullet-proof watchdog in the Python API that will ensure the VHDL test never hangs indefinitely.

My concern is whether we can implement anything in Python bullet-proof enough to handle situations with multiple threads and "unexpected" `Ctrl+C`, which is precisely the corner case where we are finding these issues. I feel we might easily end with something that is slightly more robust than the VHDL, but still not completely robust.

> I guess I don't see the value in making an external script as this is something that has to be run manually.

I envision it as replacing "python". E.g. `timed 30s -- run.py -v -g`.

> My problem with an external watchdog for the group of tests are: the test time can vary drastically with system load, adding more testbenches/parameterized tests can exceed this limit unknowingly,

Grouping tests is an option, not a requirement:

```sh
duration="$1"
shift
shift

run_watched () {
  python $@
  # get id of the previous process/task
  # sleep for $duration
  # force kill
}

if [ $# -eq 0 ]; then
  for item in `python run.py -l`; do
	run_watched "$item"
  done
else
  run_watched "$@"
fi
```

> or you're spending development time externally to the API to set something up to work on a test-by-test basis. 

This is a fair point.

> I'm not sure if Pytest (which is what I use to run all my tests) has something for this already.

I am not aware of such feature. We do use pytest in this repo to run all the tests. However, I saw some strange behaviour in ghdl-cosim. Unlike here, in ghdl-cosim pytest is used with option `-v`. It turns out that executing makefiles or shell scripts works as expected, but executing VUnit run scripts makes logs disappear. See https://github.com/ghdl/ghdl-cosim/runs/811611110?check_suite_focus=true#step:4:174. Hence, I'm not sure it's the best solution to handle tests that might get blocked.

> I think a single function call when instancing the VUnit object makes more sense then adding an external interface that needs to be set up in multiple places.

> In the Python API, it would be nice to have something like set_test_timeout(), it sets the timeout limit for all individual tests. When added to individual tests, it would allow the user to change the maximum test time for specific tests.

Yes. As other features, if this is implemented in the Python API I believe it should be available at different levels (object, lib, tb, test...). I guess the starting point would be to add a hardcoded time limit somewhere in https://github.com/VUnit/vunit/blob/master/vunit/test/runner.py#L110-L127.

LarsAsplund, 2020-06-27:
What about setting the limit for the maximum number of delta cycles? At least Modelsim used to have that option.

eine, 2020-07-13:
@GlenNicholls see [pytest-dev/pytest-timeout](https://github.com/pytest-dev/pytest-timeout).

GlenNicholls, 2020-07-13:
@eine I'm testing with decorators now! When I get some time to create the PR for this issue, I'll include that as an example. Should I put the Pytest examples in `vunit/examples/pytest` or somewhere else?

@LarsAsplund 
>What about setting the limit for the maximum number of delta cycles? At least Modelsim used to have that option.

I think that solution is okay for simulators that support a flag for this, but it seems prone to error since changing a test slightly could cause unexpected failures if the delta cycle limit is already too tight. I think delta cycles would be good because it would be a bullet-proof VHDL timeout, however, it is not intuitive and would require a lot of trial and error to get a feel for a reasonable threshold. IMO, the Python layer is the most appropriate. I'm testing with eine's suggestion for pytest-timeout and will include some examples in the PR.

eine, 2020-07-13:
> I'm testing with decorators now!

Nice! I run some quick tests on MSYS2, while trying to manage building multiple packages. For that use case, it didn't work as expected, because the commands seem to ignore the termination signal/event. However, it might work for VUnit.

Note that you need to install `pytest-timeout` explicitly. Otherwise, it fails silently...

> When I get some time to create the PR for this issue, I'll include that as an example. Should I put the Pytest examples in vunit/examples/pytest or somewhere else?

I think it would fit in that example. However, instead of naming it `pytest`, I'd name it something related to "integration", "management" of "multiple" run scripts. You can add a pytest subdir if you want.

IMHO, that example should contain one or multiple "orchestration" structures/scripts which we want VUnit to work with. One of them is using pytest, as you are doing. Another one might be @ktbarrett's work in #651. If that example grows too large, we can always split it to another repo in the future.

GlenNicholls, 2020-07-13:
@eine 
>Nice! I run some quick tests on MSYS2, while trying to manage building multiple packages. For that use case, it didn't work as expected, because the commands seem to ignore the termination signal/event. However, it might work for VUnit.

I used it in MINGW and it seems to work fine.

>I think it would fit in that example. However, instead of naming it pytest, I'd name it something related to "integration", "management" of "multiple" run scripts. You can add a pytest subdir if you want.

Okay, I kind of like integration/pytest, so I'll go with that.

eine, 2020-08-03:
@GlenNicholls, a different approach, based on Python only: https://github.com/msys2/msys2-devtools/blob/e835e769590bb78ef8d2f66708f863c70c65540e/autobuild.py#L30-L36