VUnit/vunit #663: Create Pytest example PR

GlenNicholls, 2020-07-02:
Add a PR with the following discussion [July 1, 2020 11:03 AM](https://gitter.im/VUnit/vunit?at=5efcc1f5c223cc536a49391d). This example likely solves the problem in https://github.com/VUnit/vunit/pull/558

I'm looking at https://github.com/VUnit/vunit/tree/master/tests/acceptance which is what I followed to come up with that code when I originally implemented my framework. Is it still appropriate to add an example in https://github.com/VUnit/vunit/tree/master/examples/vhdl?

eine, 2020-07-03:
I think it is appropriate to add an example because the usage of pytest in this project is non-trivial. Apart from acceptance tests (which includes +10 tests), many other pytest and run scripts are distributed along the repo. Hence, it would be useful to have a simpler example which combines just a few (a couple) scripts with bare minimal VHDL sources. It has three possible uses:

- An introduction for users that want to understand how VUnit, VHDL libs or vcomponents are tested; before jumping to the more complex scripts.
- A playground for other users to propose different combination/import strategies (e.g. #558 or https://stackoverflow.com/questions/42980036/how-to-combine-multiple-vunit-run-py-files-into-a-single-vunit-run).
- A playground to explore #662.

bryankerr1986, 2023-08-18:
Bump: Can we show an example of how to run a vunit test using pytest?