VUnit/vunit #669: Overload get() function in dictionary to return correct type

GlenNicholls, 2020-07-10:
The `get()` function with a default value in the dictionary package should be overloaded to return the correct type, the function below https://github.com/VUnit/vunit/blob/b06474f7758580e2a40e936017975ca0a1b26073/vunit/vhdl/dictionary/src/dictionary.vhd#L33-L37

to do this, it should have bodies for all standard types like so:

```
  impure function get (
    d             : frozen_dictionary_t;
    key           : string;
    default_value : real)
    return real is
  begin
    return real'value(get(d, key, to_string(default_value)));
  end function get;
```

IMO, if there is a default value, it should know what the return type is, and using `to_string` should allow correct behavior for all of the VHDL types along with VUnit types.

GlenNicholls, 2020-10-31:
I think that there should also be overloads for vhdl2008 `boolean_vector`, `time_vector`, `real_vector`, and `integer_vector` as well. However, those would require creating new files with the 2008 VUnit filename convention.

GlenNicholls, 2020-10-31:
@eine what're your thoughts about this issue? Do you think there is anything else I should include in this work or anything we should discuss?

eine, 2020-11-11:
Overall, LGTM. However, I'm not familiar with the codebase of dictionaries. Let's let Lars comment.

> For the `get()` with default parameter, should I also create aliases `get_<type>()`?

I don't get to see what value do aliases provide. Isn't it already implicit by the type of the value?

GlenNicholls, 2020-11-11:
That's true, but if you do the following for example:

```
get(dict, "signed", "10110");
```

Then this is unresolved. It could be solved with `signed'("10110")`, but I just followed the style in `queue_pkg` as it has aliases in there for `push()` and `pop()`.

GlenNicholls, 2020-11-11:
One of the problems I'm experiencing while testing is the following:

I'm overloading get() to infer the type based on the default_value. However, I'm running into problems with the following code:

```
        passed := get(test_dict, "character", '=') = '=';
        check(passed, "Expected ""="" when getting character key from test dictionary (got """ & get(test_dict, "character", '=') & """).");
```

This spits out the following error which I'm having trouble deciphering:

```
C:\Users\Glen.Nicholls\Desktop\vunit_fork\vunit\vhdl\dictionary\vunit_out\test_output\lib.tb_dictionary.Test_that_get_wit59c0a63e312bb43b9bcf6e43792da84c3df88d27\ghdl\tb_dictionary-test_fixture:error: 'value: '=' not in enumeration 'character'
```

I tried character'('='), but that gives the same error.

BTW, I defined get() like so:

```
  impure function get (
    d             : frozen_dictionary_t;
    key           : string;
    default_value : character)
    return character is
  begin
    return character'value(get(d, key, to_string(default_value)));
  end function get;
```

eine, 2020-11-12:
Which simulator are you trying that with?

GlenNicholls, 2020-11-15:
@eine I was using GHDL 5a57a31b

eine, 2020-11-15:
Are you trying to extract field 'character' from https://github.com/VUnit/vunit/blob/b06474f7758580e2a40e936017975ca0a1b26073/vunit/vhdl/dictionary/test/tb_dictionary.vhd#L31 but expect it to fail and return a default value (because the field does not exist)?

GlenNicholls, 2020-11-16:
That is correct

eine, 2020-11-17:
Did you try a bare minimal example?

```vhdl
...

  subtype frozen_dictionary_t is string;

  constant test_dict : frozen_dictionary_t := "output path : c::\foo\bar, input path : c::\ying\yang, active python runner : true"; 

  impure function get (
    d             : frozen_dictionary_t;
    key           : string;
    default_value : character)
    return character is
  begin
    return character'value(get(d, key, to_string(default_value)));
  end function get;

...
```

The issue seems to be somewhere in https://github.com/VUnit/vunit/blob/b06474f7758580e2a40e936017975ca0a1b26073/vunit/vhdl/dictionary/src/dictionary.vhd#L53-L105, isn't it?