VUnit/vunit #670: VUnit fails to parse file with commented code

PaulWozny, 2020-07-28:
Hello,

On the newest docker image  `ghdl/vunit:gcc-master` `sha256:35b8034a5102aabcd34c1d87e337b3e071eeb38d3ad23eb21cc93bd479bf1ab2` following code produces error only when commented line is in the file. 

- test.vhd
```vhd
library vunit_lib;
context vunit_lib.vunit_context;

entity mwe_parsing_comments is
    generic (runner_cfg : string);
end entity mwe_parsing_comments;

architecture beh of mwe_parsing_comments is
begin

    main : process
    begin
        test_runner_setup(runner, runner_cfg);
        test_runner_cleanup(runner);
    end process;

    -- Remove following line to compile
    --            check(false, "max5719 vunit:no response from dac");

end architecture beh;
```

- run.py
```python
from vunit import VUnit
vu = VUnit.from_argv()
testlib = vu.add_library("test")
testlib.add_source_files("test.vhd")
vu.main()
```

eine, 2020-07-28:
@PaulWozny, does it work if you remove/replace `:` in the string?

PaulWozny, 2020-08-04:
> @PaulWozny, does it work if you remove/replace `:` in the string?

Yes.

It also works when there are no characters after `:`.
For example this works as well:
```vhdl
    --            check(false, "max5719 vunit:");

```

eine, 2020-08-28:
@PaulWozny, thanks for checking! The parser, which is based on regexps, is known to work just well enough, and there are weird constraints such as this one. However, it is undesirable to make the parser more complex than it already is. See related discussions: https://github.com/VUnit/vunit/issues?q=label%3Aparsing+sort%3Aupdated-desc+

Hence, unless someone comes up with a not very complex modification to the regular expressions (https://github.com/VUnit/vunit/blob/7550dd882de2333b407637ede296bb7b26b55935/vunit/vhdl_parser.py), I think avoiding `:` in comments is the safest approach.

LarsAsplund, 2024-12-11:
See #1080

closed 2024-12-11 by LarsAsplund