VUnit/vunit #671: How to simulate with netlists?

GlenNicholls, 2020-08-04:
I have a repository with cores generated in ISE 14.7. Unfortunately, the stub files are not present so I'm unable to use XilinxCoreLib and I have to use the `.edn` netlists. ActiveHDL (and ModelSim I think?) supports this file type, but VUnit complains about the `.edn` extension:

```
Error: D:\linux_driver-trunk\Xilinx\DMA_CONTROL\src\asyncfifo32x4095v2pbr.edn(1): near "(": syntax error
```

I'm looking through the repo and I see `add_external_library()`, but this doesn't seem right since I need to add the `.edn` netlist to a specific library where said library also has `.vhd` source files. How do I go about running my VUnit tests when there are netlist files that are required to run the simulation?

GlenNicholls, 2020-08-26:
I got this from @qarlosalberto 

>ahh I think that Vunit doesn't support .edn files
https://github.com/VUnit/vunit/blob/master/vunit/source_file.py#L360
>I have seen this post: https://forums.xilinx.com/t5/Simulation-and-Verification/generate-sim-netlist/td-p/779405
>maybe you can try with write_vhdl /path/to/file.vhd it will generate a vhdl file

Still doesn't solve my problem, though. If the simulator supports `.edn` and other formats, would adding a new supported file-type fix my problem?

eine, 2020-08-28:
@GlenNicholls, the proposal would be for `add_external_library` to detect/copy/move any `*.edn` file along with `*.vhd` or `*.vhdl`? I am not sure about the details of `add_external_library`, but I think it would work.

The issue with adding netlists through regular add functions is that, depending on the simulator, each of those files might be analyzed explicitly, which would crash. If all the sources are provided at the same time, that's not an issue. However, VUnit does detect the dependencies and it supports incremental compilation. Hence, netlists should be handled as special sources attached to some parent VHDL source, and passed for analysis/elaboration only when that source needs it.

GlenNicholls, 2020-09-14:
>the proposal would be for add_external_library to detect/copy/move any *.edn file along with *.vhd or *.vhdl?

I believe so, but I have not experimented to see if using `add_external_library` would work to add a netlist to a specific library before adding the parent VHDL source via normal add. If I replicate https://github.com/VUnit/vunit/blob/e96eed6985e1f15b9afb5534ee6ba26e31efdfca/vunit/source_file.py#L360 and create something like `NETLIST_EXTENSIONS = (".edn", ".ngc"...)`, does that seem reasonable?

eine, 2020-09-14:
It does! We cannot test it in CI, tho, because GHDL does not support netlists AFAIAA. Anyway, open a PR when you try it, and we'll test it locally.