VUnit/vunit #672: Get Simulator Informations in Python API

jonasexpleo, 2020-08-10:
For testing with netlists I need some vendor libraries. These are shipped with the simulator ActiveHDL and I use this to include them
```
vu.add_external_library('pmi_work', activehdl_lattice_path / "verilog/pmi_work")
````
 In this case I need to specify the absolute path to ActiveHDL. This path is should be known by vunit, but has no stable interface to use it. My workaround is to use a private interface...
```
vu._simulator_class.find_prefix()
```

A second problem is that I need to change the path of the external library depending on the version of ActiveHDL (10.3 or 11.1).
At least [ActiveHDL](https://github.com/VUnit/vunit/blob/c418e8a18ba38313da013091495e841659ab6c05/vunit/sim_if/activehdl.py#L57) and [GHDL](https://github.com/VUnit/vunit/blob/c418e8a18ba38313da013091495e841659ab6c05/vunit/sim_if/ghdl.py#L121) has a functionality to receive the version of the program.

My idea is to define a function in the `SimulatorInterface` which would return a dict with the path and version of the simulator. Or is there a functionality which expose these information already? Any other suggestions?

eine, 2020-08-15:
Might the following sequence work for you?

`from vunit.sim_if.factory import SIMULATOR_FACTORY` (https://github.com/VHDL/Compliance-Tests/blob/b5a8580af6a9cd3b0ee40bc74a7fa02b89d0500f/run.py#L3)

https://github.com/VUnit/vunit/blob/f0d861e466eaff965357e9cbb45858e7e9a5c047/vunit/sim_if/factory.py#L106

`select_simulator` should return a class corresponding to the active/selected simulator, and from there you should be able to get the path and the version. If some of them are private, maybe we can just expose them.

@LarsAsplund, what do you think?

jonasexpleo, 2020-08-17:
Until till now I didn't though about to talk to the interface directly. It seams to be an easy solution.

I did a small PoC in my branch: [jonasexpleo/vunit](https://github.com/jonasexpleo/vunit/tree/fix672) It is only tested for GHDL and ActiveHDL, but Rivera should also work. For ModelSim I didn't find a function to receive the version information.

Probably a better unification of the version number parsing and returning would be better, but for a PoC it should be ok.

eine, 2020-08-17:
You might want to open a PR with those changes. Not meaning we should merge them (yet). But a (draft) PR will allow us to better discuss it, as we can comment on the code and keep a track of resolved issues.

GlenNicholls, 2020-08-26:
@jonasexpleo Why exactly would you need to use different versions of the same simulator? I honestly struggle with ActiveHDL (reference https://github.com/VUnit/vunit/issues/650) constantly as they do not directly test VUnit in their acceptance tests and VUnit GHA's don't have any way yet of testing against AHDL. I've always installed the newest version of AHDL and re-compiled the libraries hoping that one day they'll test enough to fix that bug since I haven't had the bandwidth at work or in my free-time to provide more debug info to solve the underlying problem in that issue.

I use multiple simulators (ModelSim, ActiveHDL, RivieraPRO, and GHDL) daily/weekly and my workaround was to define the path that my colleagues need to compile vendor libraries. My solution to part of your problem above was to create a VUnit attachment that basically couples our python build infrastructure with VUnit and I have a lot of documentation about installing and setting up all these tools for my colleagues. I haven't had a chance to find all the reference code I've posted publicly, but I've commented on other VUnit issues and in the VUnit Gitter with code about how I've solved this multi-simulator problem. Basically, my objective was to make this "attachment" an abstraction layer such that my colleagues can just call a bunch of methods to set everything up for VUnit to run instead of making all the `run.py` files deal with this complexity (didn't want to fix bugs in more than one place).

I see the value in your PR don't get me wrong, but if the above "attachment" is something you're trying to do for your company, let me know and I can be a resource and provide code-snippets. With that said, I would counter your point for the need to provide the version information of the simulator since this is really not something that needs to be known by VUnit (except if there are known bugs that break certain VUnit VHDL code) or the user. I'm sure there are cases I haven't thought about, but IMO, maintenance is easier when everything you set up in your local environment(s) is common between other users/systems. In short, my point is that for ActiveHDL, I would personally write documentation/instructions for the user to compile vendor libraries to a specific directory in the ActiveHDL so that your abstraction layer is less complex for the user.

For example, I wrote something like this:

```
        self.__SUPPORTED_SIMULATORS = ['activehdl', 'ghdl', 'modelsim', 'rivierapro']
        self.__ENVIRONMENT_VARIABLES = {
            'activehdl': 'VUNIT_ACTIVEHDL_PATH',
            'ghdl': 'VUNIT_GHDL_PATH',
            'modelsim': 'VUNIT_MODELSIM_PATH',
            'rivierapro': 'VUNIT_RIVIERAPRO_PATH'
        }
        self.__SUPPORTED_VENDOR_LIBS = {
            'xilinx': {
                'base_libs': [
                    'unisim', 'unimacro', 'unifast', 'secureip'
                ],
                'encrypted_libs': [
                    'unisim', 'unimacro', 'unifast', 'secureip', 'xpm', 'xilinx_vip'
                ],
            },
            'intel': {
                'base_libs': [
                    'altera', 'altera_lnsim', 'altera_mf', 'lpm'
                ],
                'encrypted_libs': None
            }
        }

    def __init_vendor_libs(self):
        """Initialize Vendor Libraries

        Initializes the vendor libraries by setting a private variable
        based on the vendor being targeted. This will only be run if
        encrypted libraries or core vendor libraries are enabled.
        """
        vendor_libs = (
            self.__SUPPORTED_VENDOR_LIBS[self.vendor][
                'encrypted_libs' if self.use_encrypted_libs else 'base_libs'
            ]
        )
        self.__vendor_libs = vendor_libs if self.use_vendor_libs else None

    def __init_vendor_libs_path(self):
        """Initialize Vendor Libraries Path

        Initializes the path to the location of desired vendor
        libraries
        """
        loc_to_libs = None
        if self.__simulator in ['activehdl', 'rivierapro']:
            loc_to_libs = join('..', 'vlib')
        elif self.__simulator == 'ghdl':
            loc_to_libs = join('..', 'lib', 'ghdl', 'vendors')
        elif self.__simulator == 'modelsim':
            loc_to_libs = join('..')
        self.__vendor_libs_path = join(self.__simulator_env_path, loc_to_libs)

    def add_vendor_libs(self, vunit_obj):
        """Add External Vendor Libraries

        Adds external vendor libraries to the simulator.

        For GHDL, the vendor libraries must be compiled under
        `<ghdl path>/vendors/lib` in the format that GHDL vendor library
        compile script saves them as. They should be compiled with
[... 26 lines cut ...]
            assert exists(path), f' {path} does not exist!'
            if self.__simulator == 'ghdl':
                ghdl_lib_map = {'1993': 'v93', '2008': 'v08'}
                vunit_obj.add_external_library(
                    lib_name,
                    join(path, ghdl_lib_map[self.vhdl_standard]),
                    vhdl_standard=self.vhdl_standard
                )
            else:
                vunit_obj.add_external_library(lib_name, path)
        # add XilinxCoreLib
        if self.use_xilinxcorelib:
            path = join(self.__vendor_libs_path, 'xilinx-ise', 'xilinxcorelib')
            print(f'Adding external library xilinxcorelib from {path}')
            assert exists(path), f' {path} does not exist!'
            if self.__simulator == 'ghdl':
                ghdl_lib_map = {'1993': 'v93', '2008': 'v08'}
                vunit_obj.add_external_library(
                    'xilinxcorelib',
                    join(path, ghdl_lib_map[self.vhdl_standard]),
                    vhdl_standard=self.vhdl_standard
                )
            else:
                vunit_obj.add_external_library('xilinxcorelib', path)
```

jonasexpleo, 2020-08-27:
@GlenNicholls Thank you for the detailed suggestions. First of all I like your approach. Under normal circumstances I would have used a similar way.

My current project is a safety project with hard requirements to the tool chain. To be specific we need to use Lattice Diamond in Version 3.10 with the included ActiveHDL Version 10.3 as specified in the [functional safety certificate](https://www.latticesemi.com/en/Solutions/Solutions/SolutionsDetails01/FunctionalSafety). This Lattice Edition (LE) of ActiveHDL is a cut down version the ActiveHDL suit supporting only the simulation. For the final test we need to use this specific version. For the development we have a full license of AHDL and like to keep track on the newest version (current 11.1), so we could use the performance gain and other enhancements.

To the vendor libs: in the development I could compile the libs, but for the LE I would like to use the shipped version. It is easier to argument that the certificate stills applies if the libs match the tool chain and therefore the certificate. My approach is to use in both cases the shipped vendor lib. Sadly the contained folders differs between LE and AHDL 11.1.

> With that said, I would counter your point for the need to provide the version information of the simulator since this is really not something that needs to be known by VUnit (except if there are known bugs that break certain VUnit VHDL code) or the user.

I would argument the same way, but all simulation interfaces except modelsim have build in the possibility to read out the version number. In the PR I took the only the existing implementations and defined an interface (which still needs some clean up...), so the user can use the existing information.

GlenNicholls, 2020-08-27:
>To be specific we need to use Lattice Diamond in Version 3.10 with the included ActiveHDL Version 10.3 as specified in the functional safety certificate.

Ahhh, very interesting, see I knew there were cases I did not consider!

> I would argument the same way, but all simulation interfaces except modelsim have build in the possibility to read out the version number. In the PR I took the only the existing implementations and [...]

Okay, that is good information. Out of curiosity, what about ModelSim doesn't allow you to extract the version information? Assuming the defaults are used when installing the application, this is embedded in the folder name I believe. It's been a while since I've installed ModelSim, but I seem to remember the folder name was alphabet soup with this. With that said, that is not reliable IMO and I would add something external to VUnit to gather this version info.

eine, 2020-08-28:
For context, there might be reasons to use the version of some tool as internal conditions when running regression tests. For instance, in the context of co-simulation with GHDL, not all the backends and versions support generation of binaries with embedded external C sources. For now, we discard all `mode` backends as "not suitable", however, we should improve it to e.g. accept "GHDL v0.37 mcode" and "GHDL v0.36 LLVM" but not "GHDL v0.36 mcode".

Currently, this check is not used in this repo, but in VUnit/cosim: https://github.com/VUnit/cosim/blob/master/tests/acceptance/test_examples.py#L16-L20. Moreover ghdl-cosim would benefit from taking version, backend and OS into account: https://github.com/ghdl/ghdl-cosim/blob/master/test.py#L64-L67. I think it is interesting to provide consistency between these use cases, and the ones you mentioned above.

In fact, it would not conflict but complement Glen's code example above:

```py
self.__simulator = SIMULATOR_FACTORY.select_simulator().name

...

loc_to_libs = None
if self.__simulator in ['activehdl', 'rivierapro']:
    loc_to_libs = join('..', 'vlib')
elif self.__simulator == 'ghdl':
    loc_to_libs = join('..', 'lib', 'ghdl', 'vendors')
elif self.__simulator == 'modelsim':
    loc_to_libs = join('..')
self.__vendor_libs_path = join(self.__simulator_env_path, loc_to_libs)
```