VUnit/vunit #676: Load_csv with don't care

NINI1988, 2020-09-03:
Is it some how possible to load a csv file with signal curves and check results which have don't cares?

|    |    |   |    |   | |   |
|---|---|--|---|--|--|--|
| signal curve | 0 | 1 | 1 | 0| 1 | 0|
| check value | 0 | - | 1 | -|  - | 1|

One possibility would be to set the don't care values in the csv file to `9` and make a if else, but it would be more obvious when there is a `-` in the csv file.

LarsAsplund, 2020-09-03:
No, currently we only support integers so if you need some other type it has to be encoded into an integer as you suggested. There are plans to add support for any type but that's has yet to be implemented.

LarsAsplund, 2020-09-15:
@NINI1988 Are you ok with having #521 as a placeholder for this work?

NINI1988, 2020-09-15:
Yes, that's OK.

closed 2020-09-15 by NINI1988