VUnit/vunit #678: --export-json without preprocessed path

timrid, 2020-09-14:
I noticed a problem with the tool `vunit-test-explorer` (https://github.com/Bochlin/vunit-test-explorer/issues/17). Here you can jump from the GUI to the source code of the test case with one click, which is very comfortable. 

But I use the option `vu.enable_location_preprocessing()` in my `run.py` which runs a preprocessor over the VHDL code and copys the files to `vunit_out/preprocessed/...`. When I now click on the "Show source" icon, the program always jumps to the preprocessed files in the folder `vunit_out/preprocessed/...`. If you make changes to these files, they will be overwritten on the next test execution...

The tool `vunit-test-explorer` internally uses the `--export-json` option of vunit, to collect all test cases and the files. And the `--export-json` option always prints the path to the preprocessed files.

So is it possible that the `--export-json` option additionally prints the path to the original files?

eine, 2020-09-14:
IIRC, the `--export-json` was added mainly for Sigasi (#347). I assume it is used in that context for running simulation/synthesis tools, rather than providing access to the sources. Otherwise, some other user might have noticed the same issue with `enable_location_preprocessing`.

As seen in https://github.com/VUnit/vunit/blob/8bd4064e757926b77c91de2aea68d9e26f9918ee/vunit/ui/__init__.py#L839-L894, `get_compile_order` is used for retrieving the sources.

I agree that it would be useful to provide both (the "real" sources and the processed sources), either together or as mutually exclusive lists.

/cc @davidmedinasigasi @lievenlemiengre

LarsAsplund, 2020-09-15:
@timrid Preprocessing is a bit of a pain, it often backfires. With VHDL-2019 it will be possible to get the location without preprocessing. I did some successful tests with the latest Riviera-PRO the other week and plan to integrate that into VUnit shortly. What simulator are you using?

timrid, 2020-09-15:
I work with "Active-HDL Lattice Edition 10.5.216.6767" and "vunit-hdl 4.4.0"

LarsAsplund, 2020-09-24:
Ok. Unfortunately I dont think VHDL-2019 is supported in that simulator. Can you confirm?

timrid, 2020-09-24:
That is correct. Active-HDL Lattice Edition supports only:
- VHDL 1076-1993
- VHDL 1076-2002
- VHDL 1076-2008

jevogel, 2025-03-10:
@EhabY, I agree, and this issue applies with other tools that can use the VUnit `run.py` file as a project source like the TerosHDL VS Code extension. I believe it applies with any VUnit feature that adds a preprocessing step:

- `vu.enable_location_preprocessing()`
- `vu.enable_check_preprocessing()`
- `vu.add_preprocessor()`

If you create a TerosHDL project from a VUnit `run.py`, it uses the sources listed by the VUnit `--export-json` option for the language server (VHDL-LS) and the linter (VSG or others), and these do not work correctly when the paths don't point to the original source files.

I don't think relying on some feature from VHDL-2019 to work around this issue as @LarsAsplund suggested would be sufficient as many tools have incomplete or no support for that standard, including VHDL-LS.

The issue linked in the OP https://github.com/Bochlin/vunit-test-explorer/issues/17 states that adding another field with a path to the original source file for each file would allow them to resolve the issue. I propose naming the field "source_file_name". It could look like:

```json
{
    "export_format_version": {...},
    "files": [
        {
            "file_name": "<project_path>/fpga/sim/vunit_out/preprocessed/project_lib/my_fifo.vhd",
            "source_file_name": "<project_path>/fpga/src/my_fifo.vhd",
            "library_name": "project_lib"
        },
        ...
    ],
    ...
}
```

How difficult would this be to implement?