VUnit/vunit #679: Error (suppressible): (vsim-19) Failed to access library 'work' at "work".

fpgaexcel, 2020-09-15:
Folks,

any help would be appreciated.

I am using the xpm_cdc_sync_rst module from the Xilinx XPM Library.

It references glbl.v

I have tried the following:

(1) vsim <other options> glbl
(2) I have added glbl.v to my source files that are compiled into my one "source library"
(3) I have tried to add a specific "work" library using vunit - this is not allowed

None allow the the simulation to run.

Any ideas?

Cheers!

LarsAsplund, 2020-09-15:
The reason we do not allow a library named `work` is described [WORK is not a VHDL library](https://insights.sigasi.com/tech/work-not-vhdl-library/).

Could you provide your `run.py` script?

Generally we recommend compiling third party IPs using whatever method the vendor recommends and then reference these compiled libraries using [`add_external_libraries`](http://vunit.github.io/py/vunit.html?highlight=add_external#vunit.ui.VUnit.add_external_library)

fpgaexcel, 2020-09-16:
Lars,

thanks for the response.

Python is as follows:

#!/bin/python
import os
from os.path import join, dirname
from itertools import product
from vunit.verilog import VUnit

vu = VUnit.from_argv()

src_path = join(dirname(__file__), "../rtl/")
src_lib = vu.add_library("src_lib")
src_lib.add_source_files(join(src_path, "clockReset.sv"))

tb_lib = vu.add_library("tb_lib")
tb_lib.add_source_files("*.sv")

tb_lib.set_sim_option("modelsim.vsim_flags", ["-novopt"])
vu.add_external_library("unisim", join(os.getenv('XILINX_SIMLIB_PATH'), "unisim"));
vu.add_external_library("simprims_ver", join(os.getenv('XILINX_SIMLIB_PATH'), "simprims_ver"));
vu.add_external_library("xpm", join(os.getenv('XILINX_SIMLIB_PATH'), "xpm"));
vu.set_sim_option("modelsim.init_file.gui", "wave.do")

vu.main()

Error is:

# ** Error: (vsim-3043) Unresolved reference to 'glbl'.
#    Time: 0 ps  Iteration: 0  Instance: /tb_clock_reset/clockReset/xpm_cdc_sync_rst100_inst File: /opt/Xilinx/Vivado/2019.2/data/ip/xpm/xpm_cdc/hdl/xpm_cdc.sv Line: 1119

Any help you could provide would be appreciated!

~

aodling, 2020-09-16:
Hi!

There is one solution i know of at least.

The glbl.v is a xilinx file with FPGA global signals. We usually include it
as source and in vsim_flags you add "src_lib.glbl" as a new list item.

There could be smarter ways but ;P

Regards
Andreas

fpgaexcel, 2020-09-17:
Andreas,

That was the trick, that worked for me!

Not sure if this is what you intended but I did have to explicitly compile it thus:

src_lib.add_source_files(join(ver_src_path, "glbl.v"))

followed by

tb_lib.set_sim_option("modelsim.vsim_flags", ["-novopt src_lib.glbl"])

That is a cleaner solution than mine which also worked however it was a tad naughty:

I removed the check (and subsequent Error) for a work library in vunit/project.py.....

I prefer your solution.

Cheers!

aodling, 2020-09-17:
Hi!
Glad I could help!
That was exactly how I meant it, That is how we do at my work. I don't like
to add it as a sim option but I guess it is required when working with
xilinx IP/primitives :)

Good luck testning!

/A

eine, 2020-09-17:
I'm closing this because the issue seems fixed. Thanks @aodling and @fpgaexcel for this nice interaction!

closed 2020-09-17 by eine

benthie, 2025-01-10:
Just FYI for all who might have to migrate from ModelSim to Questa (Base) using the 3-step-flow (see #899). The simulation option has to be adapted as follows:
~~~
vu.set_sim_option("modelsim.vopt_flags", ["lib.glbl"])
~~~