VUnit/vunit #687: No opportunity to append sim_option in test_case

arnfol, 2020-10-21:
Hi!
I faced this problem while trying to add an additional string in `"modelsim.vsim_flags"` sim option in different test cases. Basically, I have a default configuration for my entire testbench with `"modelsim.vsim_flags"` set. In only one test in my tb, I wanted to add some configurations, and in each configuration add a new different argument to `"modelsim.vsim_flags"`. 
However, `add_config` method in Test does not have an _overwrite_ argument, like `set_sim_option`. I was thinking about working directly with test configuration object, as it has `set_sim_option` method, but it does not support _overwrite_ argument either.

arnfol, 2020-10-21:
I added an example of what I am trying to achieve at https://github.com/malt-system/vunit/commit/1985ef2e4501a6362b02aea9b13db81e42584da5 and an ugly workaround at https://github.com/malt-system/vunit/commit/887aef36ef676c1b7b0b1af10ffc495791d84aca.
I'd like to add the _overwrite_ argument to either `add_config` method or test configuration's `set_sim_option`, or both. However, I am not sure if this is the right solution, could use a piece of advice here.

eine, 2020-11-11:
It seems that there is a single `set_sim_option` which does not accept overwrite:

https://github.com/VUnit/vunit/search?q=%22def+set_sim_option%22

I'd say that moving https://github.com/VUnit/vunit/blob/c35d3dc392ef7557db1dcbba61d86753c912c635/vunit/configuration.py#L224-L227 to https://github.com/VUnit/vunit/blob/c35d3dc392ef7557db1dcbba61d86753c912c635/vunit/configuration.py#L122-L127 would be the easiest fix.

Moreover, it might be interesting to add the overwrite feature to `add_config`. However, just fixing `set_sim_option` would already provide a consistent solution.