VUnit/vunit #688: Feature request: VHDL 93C support

DeniTCH, 2020-10-26:
Hi,

I faced the problem when I tried to use VUnit for building the Xilinx unisim library as a part of my project (GHDL). The version of the library in question comes with ISE 14.7 (the last version with support for this particular FPGA). This version of the library must be built with 93c (does not work with 93) and works fine when I invoke GHDL manually. When I realized that the default VHDL standard for VUnit was 2008, i tried to change it by setting the `vhdl_standard` parameter, but when I dug deeper I found out that 

`vhdl_standard.py` does not define the 93c standard: `_STANDARDS = {"1993", "2002", "2008", "2019"}`

I am new to VUnit and am not certain how big of an effort it is to add support for an additional standard?

eine, 2020-10-30:
Hi @DeniTCH! `93c` is not an standard revision. As you can see in https://ghdl.github.io/ghdl/using/ImplementationOfVHDL.html#vhdl-standards:

> 93c: Same as 93 and -frelaxed.

Hence, you and use 1993 in VUnit along with the following options:

```py
VU.set_compile_option("ghdl.flags", ["-frelaxed"])
VU.set_sim_option("ghdl.elab_flags", ["-frelaxed"])
```

See https://github.com/VUnit/vunit/blob/master/examples/vhdl/array/run.py#L28-L29

DeniTCH, 2020-10-30:
Hello @eine, thank you for the clarification! I am relatively new to VHDL and ghdl and was not aware of this definition. The unisim compiles fine now.

closed 2020-10-30 by DeniTCH