VUnit/vunit #689: vunit doesn't work in WSL

sepehrMSP, 2020-10-29:
I have installed Modelsim SE-64 2019.4 on my windows and I intend to run some systemverilog simulation test via vunit in WSL.

When I'm running Vunit examples for verilog, I will get following errors:

```user@DESKTOP-S7405AK:~/vunit_smaple/usergide$ python run.py
Compiling into vunit_lib: ../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv passed
Compiling into lib:       tb_example_basic.sv                                                 passed
Compiling into lib:       tb_example.sv                                                       passed
Compile passed

Starting lib.tb_example.Test that a successful test case passes
Output file: vunit_out/test_output/lib.tb_example.Test_that_a_successful_test_case_passes_2bdda5abfac164787ee41954d1c48bd036699331/output.txt
Traceback (most recent call last):
  File "/home/user/.local/lib/python2.7/site-packages/vunit/test/runner.py", line 223, in _run_test_suite
    results = test_suite.run(output_path=output_path, read_output=read_output)
  File "/home/user/.local/lib/python2.7/site-packages/vunit/test/list.py", line 107, in run
    test_ok = self._test_case.run(*args, **kwargs)
  File "/home/user/.local/lib/python2.7/site-packages/vunit/test/suites.py", line 72, in run
    results = self._run.run(*args, **kwargs)
  File "/home/user/.local/lib/python2.7/site-packages/vunit/test/suites.py", line 187, in run
    sim_ok = self._simulate(output_path)
  File "/home/user/.local/lib/python2.7/site-packages/vunit/test/suites.py", line 258, in _simulate
    elaborate_only=self._elaborate_only,
  File "/home/user/.local/lib/python2.7/site-packages/vunit/sim_if/vsim_simulator_mixin.py", line 340, in simulate
    return self._run_persistent(common_file_name, load_only=elaborate_only)
  File "/home/user/.local/lib/python2.7/site-packages/vunit/sim_if/vsim_simulator_mixin.py", line 301, in _run_persistent
    self._persistent_shell.execute('source "%s"' % fix_path(common_file_name))
  File "/home/user/.local/lib/python2.7/site-packages/vunit/persistent_tcl_shell.py", line 62, in execute
    process.writeline(cmd)
  File "/home/user/.local/lib/python2.7/site-packages/vunit/ostools.py", line 154, in writeline
    self._process.stdin.write(line + "\n")
IOError: [Errno 32] Broken pipe
fail (P=0 S=0 F=1 T=4) lib.tb_example.Test that a successful test case passes (3.0 seconds)
```
I have made simlinks to all .exe files in the `win64` directory of modelsim installation path. and Vunit version is 4.3.0 .
I don't know what's the problem. can you help me?

eine, 2020-10-30:
You are using Python 2.7, which is deprecated since 2020/01/01. Use Python >=3.6 instead.

sepehrMSP, 2020-10-30:
I've run it again with python 3.8, but the problem is still there (note that I'm running usesrgide sample for verilog in the repo). see the errors:
```
user@DESKTOP-S7405AK:~/vunit_smaple/usergide$ python3 run.py
Compiling into vunit_lib: ../../.local/lib/python3.8/site-packages/vunit/verilog/vunit_pkg.sv passed
Compiling into lib:       tb_example_basic.sv                                                 passed
Compiling into lib:       tb_example.sv                                                       passed
Compile passed

Starting lib.tb_example.Test that a successful test case passes
Output file: vunit_out/test_output/lib.tb_example.Test_that_a_successful_test_case_passes_2bdda5abfac164787ee41954d1c48bd036699331/output.txt
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.8/site-packages/vunit/test/runner.py", line 222, in _run_test_suite
    results = test_suite.run(output_path=output_path, read_output=read_output)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/test/list.py", line 107, in run
    test_ok = self._test_case.run(*args, **kwargs)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/test/suites.py", line 72, in run
    results = self._run.run(*args, **kwargs)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/test/suites.py", line 187, in run
    sim_ok = self._simulate(output_path)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/test/suites.py", line 254, in _simulate
    return self._simulator_if.simulate(
  File "/home/user/.local/lib/python3.8/site-packages/vunit/sim_if/vsim_simulator_mixin.py", line 343, in simulate
    return self._run_persistent(str(common_file_name), load_only=elaborate_only)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/sim_if/vsim_simulator_mixin.py", line 301, in _run_persistent
    self._persistent_shell.execute('source "%s"' % fix_path(common_file_name))
  File "/home/user/.local/lib/python3.8/site-packages/vunit/persistent_tcl_shell.py", line 61, in execute
    process.writeline(cmd)
  File "/home/user/.local/lib/python3.8/site-packages/vunit/ostools.py", line 146, in writeline
    self._process.stdin.write(line + "\n")
BrokenPipeError: [Errno 32] Broken pipe
fail (P=0 S=0 F=1 T=4) lib.tb_example.Test that a successful test case passes (2.3 seconds)
```
What should I do now?

eine, 2020-10-31:
Which version of WSL2 and which container are you using?

sepehrMSP, 2020-10-31:
See below information:
```
C:\Users\user>wsl -l -v
  NAME      STATE           VERSION
* Ubuntu    Running         1

C:\Users\user>ver

Microsoft Windows [Version 10.0.19041.572]
```
Therefore it seems to be 1, How ever there are some answers on the internet which indicates that , if your windows build version is greater than 18917 you are probably using WSL2. 
Also some answers mentioned that if your kernel version is => 4.19, WSL version is 2. See below (it's on ubuntu shell):
```
user@DESKTOP-S7405AK:~$ uname -a
Linux DESKTOP-S7405AK 4.4.0-19041-Microsoft #488-Microsoft Mon Sep 01 13:43:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Actually I'm confused, but I think you can find out the version with these information. And, what do you mean by container? I've searched for it, but most of titles were related to Docker while I'm not using it.
 Now, what is the next step?

eine, 2020-11-11:
By "constainer" I meant the WSL image. Anyway, let's move the discussion to #658.

closed 2020-11-11 by eine