VUnit/vunit #701: What child process is VUnit being unable to create?

curiousengineer, 2020-12-01:
Here is the problem:

$ make compile
Traceback (most recent call last):
  File "C:\My_Designs\probe_fpga_design_1\run.py", line 336, in <module>
    main()
  File "C:\My_Designs\probe_fpga_design_1\run.py", line 181, in main
    vu.add_osvvm()
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\ui\__init__.py", line 1030, in add_osvvm
    self._builtins.add("osvvm")
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\builtins.py", line 44, in add
    self._builtins_adder.add(name, args)
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\builtins.py", line 280, in add
    function(**args)
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\builtins.py", line 161, in _add_osvvm
    self._simulator_class.supports_vhdl_package_generics()
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\sim_if\activehdl.py", line 61, in supports_vhdl_package_generics
    proc = Process(
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\ostools.py", line 104, in __init__
    self._process = subprocess.Popen(
  File "c:\users\user_111\appdata\local\programs\python\python39\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\users\user_111\appdata\local\programs\python\python39\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
Exception ignored in: <function Process.__del__ at 0x000001F43A66C4C0>
Traceback (most recent call last):
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\ostools.py", line 242, in __del__
    self.terminate()
  File "c:\my_designs\probe_fpga_design_1\deps\vunit\vunit\ostools.py", line 215, in terminate
    if self._process.poll() is None:
AttributeError: 'Process' object has no attribute '_process'
make: *** [Makefile:21: compile] Error 1

The machine had ActiveHDL 10.5a 32-bit. Then ActiveHDL 11.1 64-bit was installed. The PATH to the 64-bit version was first moved up in order of precedence to force VUnit to use it. This did not make VUnit use the 64-bit version. Then, the PATH to the 32-bit version was deleted. This led to a situation where the design was being automatically compiled in 32-bit ActiveHDL 10.5a but when simulation is started, it would start the ActiveHDL 11.1 64-bit which would then complain that the compiled files are not compatible with it and simulation would not proceed. Eventually the ActiveHDL 10.5a was deleted from the computer, this led to the above error.

What child process is VUnit being unable to create here?

eine, 2020-12-02:
What is the context? That is:

- What does `$ make compile` do?
- Which terminal are you executing the command on?
- What are the versions of Python and VUnit?

curiousengineer, 2020-12-02:
The make compile basically invokes the run.py.

I have found why this error occurs. When the ActiveHDL 10.5a was uninstalled, the PATH to it was not deleted. The folder C:\Aldec\Active-HDL-10.5a also was not deleted although its contents were removed by the uninstaller program. VUnit assumed that the Active-HDL-10.5a is still present at tried to run the avhdl.exe which failed. Apparently, VUnit does not expect this type of scenario and thus does not have a built in way to tackle it. VUnit is able to tackle a situation where it does not find any simulator on the system but not one where a folder for it exist but the executable does not.

I did try to move path to the C:\Aldec\Active-HDL-11.1-x64, higher in the PATH list but this did not help. Eventually, I deleted all entries in PATH to any ActiveHDL simulator and created the environment variables VUNIT_SIMUALTOR which was set to activehdl, and VUNIT_ACTIVEHDL_PATH which was set to C:\Aldec\Active-HDL-11.1-x64. Now the above error does not occur anymore.

I am able to reproduce the problem easily now. Basically, C:\Aldec\Active-HDL-10.5a-x64 exists but is virtually empty and the C:\Aldec\Active-HDL-11.1-x64 is my preferred simulator now. If I set the VUNIT_ACTIVEHDL_PATH to C:\Aldec\Active-HDL-10.5a-x64 , I get the error shown in the original post. If I set this variable to C:\Aldec\Active-HDL-11.1-x64 again, the error goes away and everything starts to work fine. This error will occur even if the C:\Aldec\Active-HDL-10.5a-x64 is deleted. 

This leads me to the conclusion that VUnit blindly follows the environment variable path to the simulator, whether it be inside the PATH or inside VUNIT_ACTIVEHDL_PATH and performs no further checks to see if the path is valid. It would be great if VUnit can produce a more easy to understand error message in this type of scenario.

curiousengineer, 2021-01-13:
As long as VUnit tests that a simulator directory actually contains the simulator executables before opening a subprocess, this problem will be solved.