VUnit/vunit #702: Memory Model with AvalonMM Slave Breaks with Data Widths > 8 bits and Multiple Bursts

manjagu, 2020-12-07:
I created a simple project to burst data into an Avalon MM slave with memory. This is based on the tb_avalon_slave.vhd unit test in VUnit 4.4.0. I expanded the reads and writes to burst a series of numbers into the interface multiple times and then read them back. What I see is the read-backs don't work consistently (see image below as an example). I have created project files for reference at https://github.com/manjagu/avalonmm_slave.git.

My setup:
VUnit 4.4.0
Python 3.8.5
Modelsim DE 2019.2

[image]

slaweksiluk, 2020-12-09:
https://github.com/manjagu/avalonmm_slave/blob/238220ccc09c30310b6f3b32bf39071fbb2fded8/tb_avalonmm_slave.vhd#L49 this sholud be 
```vhdl
  constant MEM_SIZE_BYTES        : integer := tb_cfg.burst_count *(tb_cfg.data_width / 8.0) * tb_cfg.cycles;
```
after fix test still fails. Probably due to:
```
https://github.com/manjagu/avalonmm_slave/blob/238220ccc09c30310b6f3b32bf39071fbb2fded8/tb_avalonmm_slave.vhd#L175
```
byte address is incremented, while 16bit word used.

As far as I remember Avalon spec permits user defined address granularity - I have to check it. Vunit Avalon Slave VC support only byte addressing.

manjagu, 2020-12-09:
I think #L49 is a red herring. It was written so that if I were to use a non-multiple of 8 for a data width that enough memory would be allocated. Being that this example only uses multiples of 8, this is not an issue. Please see: 
`https://github.com/manjagu/avalonmm_slave/blob/main/run.py#L46`

I don't think line 175 is an issue either. The address specified by 'i' is appropriately re-sized to mem_address using to_unsigned. A "byte address" should not be relevant, only a word address is. For confirmation of this, please compare this to the VUnit distribution test bench code:

`https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/test/tb_avalon_slave.vhd#L48`
and
`https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/test/tb_avalon_slave.vhd#L38`

In the VUnit test code the address is incremented as a 32-bit word and the data width is also 32-bits. If your statement were correct, then the test bench for the avalon slave is not representative of actual functionality. I think the test bench for the Avalon slave doesn't include enough test cases and should be expanded to multiple burst writes and reads to see this error.

This is born out by copying and updating the lines: `https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/test/tb_avalon_slave.vhd#L114-#L128`

and inserting them at line 130. My code continues the number sequence and updates the address. My implementation below:

```vhdl
      for i in 0 to tb_cfg.num_cycles-1 loop
        if i = 0 then
          address <= std_logic_vector(to_unsigned(tb_cfg.num_cycles, address'length));
          burstcount <= std_logic_vector(to_unsigned(tb_cfg.num_cycles, burstcount'length));
        end if;
        write <= '1';
        writedata <= std_logic_vector(to_unsigned(i+tb_cfg.num_cycles, writedata'length));
        wait until rising_edge(clk) and waitrequest = '0';
      end loop;
      write <= '0';
      address <= (others => 'U');
      burstcount <= (others => 'U');
      writedata <= (others => 'U');      

      wait until rising_edge(clk);
```

I have not fundamentally altered the unit test, but added another burst transaction. The read cycle remains the same. However, the output is incorrect:
[image]

slaweksiluk, 2020-12-10:
I remember I had some issues with address granularity in avalon. I have to get closer look on that.

manjagu, 2020-12-11:
@slaweksiluk were you able to reproduce the issue using the VUnit unit test modification I mentioned above?

slaweksiluk, 2020-12-14:
@manjagu sorry I'm short of time to test it now, but I will do it in a few days

manjagu, 2021-01-05:
@slaweksiluk I've been experimenting a little further. I notice odd behavior with the memory package. Please have a look at this code and it's output. I am simply writing to memory and reading it back. Is the first delay expected?
```vhdl

      for i in 0 to tb_cfg.burst_count-1 loop
        test_signal_write_data <= std_logic_vector(to_unsigned(i, 8));
        write_word(memory, i, test_signal_write_data);
        wait until rising_edge(clk_100);        
        test_signal_read_data <= read_word(memory, i, 1);
        wait until rising_edge(clk_100);        
      end loop;
```
[image]

manjagu, 2021-01-06:
@slaweksiluk and @LarsAsplund I believe the issue is with the memory_pkg. I've started diving into debugging it, but ran out of time today. I uploaded a new series of tests in this repository: [https://github.com/manjagu/vunit_memory_pkg_expt.git](https://github.com/manjagu/vunit_memory_pkg_expt.git). The short of it is, when I write and read in the same loop, I have no issues. When I separate the write and read out into different loops, I get bad data from memory_pkg. 

To start, please have a look at the output of run.py. The fourth test fails for me. As always, if you see something I am doing incorrectly, please let me know. Thanks!

slaweksiluk, 2021-01-06:
There are two problems
* Be aware of delta cycles delay when using signals in process. In vhdl signals get new value in next delta cycle (while variables get is immediately). This is fixed by two `wait for 0 ps` lines. Hint: use variables instead. 
* Vunit memory model use byte address. Fixed by referencing memory with `i*MEM_DATA_WIDTH_BITS/8`.
```
diff --git a/tb_memory.vhd b/tb_memory.vhd
index 8a62b0b..84d29f1 100644
--- a/tb_memory.vhd
+++ b/tb_memory.vhd
@@ -99,11 +99,13 @@ begin
       for i in 0 to tb_cfg.cycles-1 loop
         test_address <= std_logic_vector(to_unsigned(i, MEM_ADDR_WIDTH)); -- for sim viewing only
         test_signal_write_data <= std_logic_vector(to_unsigned(i, MEM_DATA_WIDTH_BITS));
+        wait for 0 ps;^M
         test_signal_write_data_z1 <= test_signal_write_data; -- this was done for simulator viewing; yes, variables could have been used
         wait until rising_edge(clk_100);
-        write_word(memory, i, test_signal_write_data);
-        test_signal_read_data <= read_word(memory, i, MEM_DATA_WIDTH_BITS/8);
-        check_equal(test_signal_read_data, test_signal_write_data_z1);
+        write_word(memory, i*MEM_DATA_WIDTH_BITS/8, test_signal_write_data);^M
+        test_signal_read_data <= read_word(memory, i*MEM_DATA_WIDTH_BITS/8, MEM_DATA_WIDTH_BITS/8);^M
+        wait for 0 ps;^M
+        check_equal(test_signal_read_data, test_signal_write_data_z1, "index"&to_string(i));^M
       end loop;
 
     -----------------------------------------------------------------------------
@@ -118,7 +120,7 @@ begin
         test_address <= std_logic_vector(to_unsigned(i, MEM_ADDR_WIDTH)); -- for sim viewing only
         test_signal_write_data <= std_logic_vector(to_unsigned(i, MEM_DATA_WIDTH_BITS));
         wait until rising_edge(clk_100);
-        write_word(memory, i, test_signal_write_data);
+        write_word(memory, i*MEM_DATA_WIDTH_BITS/8, test_signal_write_data);^M
       end loop;
 
       wait until rising_edge(clk_100);
@@ -127,7 +129,7 @@ begin
       info(tb_logger, "Reading...");
       for i in 0 to tb_cfg.cycles-1 loop
         test_address <= std_logic_vector(to_unsigned(i, MEM_ADDR_WIDTH)); -- for sim viewing only
-        test_signal_read_data <= read_word(memory, i, MEM_DATA_WIDTH_BITS/8);
+        test_signal_read_data <= read_word(memory, i*MEM_DATA_WIDTH_BITS/8, MEM_DATA_WIDTH_BITS/8);^M
         wait until rising_edge(clk_100);
         check_equal(test_signal_read_data, i);
       end loop;
```

It still does not mean there is no bug in avalon mm - I haven't investigated it exactly. The first step is, as you mentioned, to add test cases for multiple data widths in current vunit avalon mm tb to reproduce bug, You can open merge request for it - I will review.

manjagu, 2021-01-10:
@slaweksiluk Thank you. The addressing by bytes was the issue with the memory VCI. I forgot about that. The AvalonMM memory I/F has the same requirements. I made the changes to fix the address and all pass. Please note, the rest of the code works as intended and the "wait for 0 ps" statements are not necessary. I will continue my investigation with the AvalonMM VCI now.

manjagu, 2021-01-11:
@slaweksiluk I confirmed the issue was entirely confusion on my part about the AvalonMM address granularity. Thank you for your help.

closed 2021-01-11 by manjagu