VUnit/vunit #705: Add missing VUunit macros for ActiveHDL simulator when running in GUI mode

curiousengineer, 2021-01-13:
When a simulator runs in GUI mode, the following procedures are expected to be available from the GUI console window:

```
# vunit_help
#   - Prints this help
# vunit_load [vsim_extra_args]
#   - Load design with correct generics for the test
#   - Optional first argument are passed as extra flags to vsim
# vunit_user_init
#   - Re-runs the user defined init file
# vunit_run
#   - Run test, must do vunit_load first
# vunit_compile
#   - Recompiles the source files
# vunit_restart
#   - Recompiles the source files
#   - and re-runs the simulation if the compile was successful
```

These are defined inside TCL files that are automatically generated by VUnit. These files are "sourced" into the simulator once it starts in GUI mode. The script files are called gui.tcl, common.tcl and batch.tcl.

The existing version of VUnit does not define all of these macros for ActiveHDL. They are all supposed to work in ModelSim and RivieraPro according to my current understanding. The reason for not defining all of them is that ActiveHDL console is not a true TCL console. This is partially true in that by default the console is not in TCL mode. This can be changed by executing scripterconf -tcl command. This shall change the console mode into a true TCL console.

At present vunit_run and vunit_load are defined inside the common.tcl. They can be run from the console window once it has been put into TCL mode by executing the scripterconf -tcl command. It should be possible to describe the rest of the commands vunit_user_init, vunit_compile and vunit_restart for ActiveHDL also.

The script generation happens inside activehdl.py for ActiveHDL. The details of scripterconf can be found inside user manual of ActiveHDL.

curiousengineer, 2021-01-13:
Here is the user manual excerpt for the scripterconf command. It has been taken from the user manual of the ActiveHDL 11.1 64x.
[scripterconf command.pdf](https://github.com/VUnit/vunit/files/5809216/scripterconf.command.pdf)

maleiter, 2024-06-13:
Any updates on this issue?

LarsAsplund, 2024-06-14:
@curiousengineer @maleiter I made a start on this. See if it works with your Aldec version.

curiousengineer, 2024-06-18:
I have moved on to different roles now. If anyone knows how to do
compilation and restart a test from within ActiveHDL then kindly let me
know.

maleiter, 2024-06-19:
> @curiousengineer @maleiter I made a start on this. See if it works with your Aldec version.

Perfect, I've tested it and it works as expected.

LarsAsplund, 2024-06-19:
Thanks @maleiter! I will merge it.

commit 9ad44451 (2024-06-14): Added vunit_help, vunit_compile, and vunit_restart commands to Active-HDL. Closes #705.
commit 70e101c9 (2024-06-15): Added vunit_help, vunit_compile, vunit_restart, and vunit_user_init commands to Active-HDL. Closes #705.
commit 6d79088d (2024-06-19): Added vunit_help, vunit_load, vunit_run, vunit_compile, vunit_restart, and vunit_user_init TCL commands to Active-HDL. Closes #705.
commit de2d4773 (2024-06-19): Added vunit_help, vunit_load, vunit_run, vunit_compile, vunit_restart, and vunit_user_init TCL commands to Active-HDL. Closes #705.
closed 2024-06-21 by LarsAsplund